ros2/rcl #792: test_logging_rosout test failing on macOS

azeey, 2020-09-04:
## Bug report

**Required Info:**

- Operating System:
  - macOS Mojave
- Installation type:
  - source
- Version or commit hash:
  - master since 2020-09-03
- DDS implementation:
  - RTI Connext

#### Steps to reproduce issue
The test TestLoggingRosoutWithDifferentSettings is failing on macOS repeated CI with the following log
```
[ RUN      ] TestLoggingRosoutWithDifferentSettings/TestLoggingRosoutFixture__rmw_connext_cpp.test_logging_rosout/test_enable_explicit_global_rosout_enable_node_option
[INFO] [1599124760.684844240] [ns.test_rcl_node_logging_rosout]: SOMETHING
/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rcl/rcl/test/rcl/test_logging_rosout.cpp:190: Failure
Expected equality of these values:
  success
    Which is: false
  GetParam().expected_success
    Which is: true
[  FAILED  ] TestLoggingRosoutWithDifferentSettings/TestLoggingRosoutFixture__rmw_connext_cpp.test_logging_rosout/test_enable_explicit_global_rosout_enable_node_option, where GetParam() = test_enable_explicit_global_rosout_enable_node_option (4331 ms)
```

The test [history](https://ci.ros2.org/view/nightly/job/nightly_osx_repeated/lastCompletedBuild/testReport/projectroot/test/test_logging_rosout__rmw_connext_cpp/history/) shows that this might be a flaky test.

#### Expected behavior
test passes

#### Actual behavior
test fails

jacobperron, 2020-09-24:
This hasn't occurred on my watch, but I wouldn't be surprised if it's flaky given the test depends on waiting for a subscription to be ready.

ivanpauno, 2020-10-14:
Connext discovery is sometimes slow, I would extend the 10 retries to 50 (max wait time goes from 1s to 5s).

ivanpauno, 2020-10-28:
This is still failing sometimes, I will increase the number of retries to see if that makes the test less flaky.

closed 2020-10-28 by ivanpauno