ros2/rcl #815: Improve error message for malformed YAML files

s-hillerk, 2020-09-29:
## Bug report

Passing a malformed YAML parameter file crashes nodes without hinting at the cause. The symptoms are the same as #553:
```
free(): double free detected in tcache 2
Aborted (core dumped)
```

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - ROS2 deb repo via apt
- Version or commit hash:
  - ros-foxy-rcl 1.1.7-1focal.20200921.18150
- DDS implementation:
  - default
- Client library (if applicable):
  - (rclpy)

#### Steps to reproduce issue

```
echo malformed::: > /tmp/settings.yaml
ros2 run <any node> --ros-args --params-file /tmp/settings.yaml
```

#### Expected behavior
The user is informed that the YAML file is malformed.

#### Actual behavior
The program crashes. The user is puzzled and is not told the cause of the crash.

clalancette, 2020-10-05:
I think this is essentially a request to backport #556 to Foxy (since I confirmed that it is fixed on the latest).  I'll add this to the Foxy board for a backport.

clalancette, 2020-10-05:
Help doing this would be appreciated.

jacobperron, 2020-10-09:
With the latest release `1.1.8`, we don't see a double free error, but instead:

```
[ERROR] [1602279724.674172361] [rcl]: Failed to parse global arguments

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'Couldn't parse params file: '--params-file /tmp/settings.yaml'. Error: Cannot have a value before ros__parameters at line 1, at /tmp/binarydeb/ros-foxy-rcl-yaml-param-parser-1.1.8/src/parse.c:623, at /tmp/binarydeb/ros-foxy-rcl-1.1.8/src/rcl/arguments.c:391'

with this new error message:

  'context is zero-initialized, at /tmp/binarydeb/ros-foxy-rcl-1.1.8/src/rcl/context.c:51'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [1602279724.674250769] [rclcpp]: failed to finalize context: context is zero-initialized, at /tmp/binarydeb/ros-foxy-rcl-1.1.8/src/rcl/context.c:51
terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
  what():  failed to initialize rcl: error not set
```

Which has a better error message, though we could probably improve things so that we avoid overwriting the error state.

jacobperron, 2020-10-19:
I'm going to close this since it looks like the error message has been improved since the latest release for Foxy.

closed 2020-10-19 by jacobperron