ros2/rcl #852: 👩‍🌾 test_info_by_topic__rmw_connext_cpp regression on Linux repeated

chapulina, 2020-11-02:
The test failed during the latest build, on 2020-11-02:

https://ci.ros2.org/view/nightly/job/nightly_linux_repeated/2095/testReport/projectroot/test/test_info_by_topic__rmw_connext_cpp/history/

Failure output:

```
[ RUN      ] TestInfoByTopicFixture__rmw_connext_cpp.test_rcl_get_subscriptions_info_by_topic_null_topic

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'topic_name argument is null, at /home/jenkins-agent/workspace/nightly_linux_repeated/ws/src/ros2/rcl/rcl/src/rcl/graph.c:456'

with this new error message:

  'failed to create participant, at /home/jenkins-agent/workspace/nightly_linux_repeated/ws/src/ros2/rmw_connext/rmw_connext_shared_cpp/src/node.cpp:316'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
RTI Data Distribution Service Evaluation License issued to OSRF ... For non-production use only.
Expires on 08-oct-2021 See www.rti.com for more information.
[D0108|ENABLE]DDS_DomainParticipant_enableI:Automatic participant index failed to initialize. PLEASE VERIFY CONSISTENT TRANSPORT / DISCOVERY CONFIGURATION. 
DDSDomainParticipant_impl::createI:ERROR: Failed to auto-enable entity
DomainParticipantFactory_impl::create_participant():!create failure creating participant
/home/jenkins-agent/workspace/nightly_linux_repeated/ws/src/ros2/rcl/rcl/test/rcl/test_info_by_topic.cpp:71: Failure
Expected equality of these values:
  0
  ret
    Which is: 1
failed to create participant, at /home/jenkins-agent/workspace/nightly_linux_repeated/ws/src/ros2/rmw_connext/rmw_connext_shared_cpp/src/node.cpp:316, at /home/jenkins-agent/workspace/nightly_linux_repeated/ws/src/ros2/rcl/rcl/src/rcl/node.c:261
```

and

```
[ RUN      ] TestInfoByTopicFixture__rmw_connext_cpp.test_rcl_get_subscriptions_info_by_topic_null_topic

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'topic_name argument is null, at /home/jenkins-agent/workspace/nightly_linux_repeated/ws/src/ros2/rcl/rcl/src/rcl/graph.c:456'

with this new error message:

  'failed to create participant, at /home/jenkins-agent/workspace/nightly_linux_repeated/ws/src/ros2/rmw_connext/rmw_connext_shared_cpp/src/node.cpp:316'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
RTI Data Distribution Service Evaluation License issued to OSRF ... For non-production use only.
Expires on 08-oct-2021 See www.rti.com for more information.
[D0108|ENABLE]DDS_DomainParticipant_enableI:Automatic participant index failed to initialize. PLEASE VERIFY CONSISTENT TRANSPORT / DISCOVERY CONFIGURATION. 
DDSDomainParticipant_impl::createI:ERROR: Failed to auto-enable entity
DomainParticipantFactory_impl::create_participant():!create failure creating participant
/home/jenkins-agent/workspace/nightly_linux_repeated/ws/src/ros2/rcl/rcl/test/rcl/test_info_by_topic.cpp:71: Failure
Expected equality of these values:
  0
  ret
    Which is: 1
failed to create participant, at /home/jenkins-agent/workspace/nightly_linux_repeated/ws/src/ros2/rmw_connext/rmw_connext_shared_cpp/src/node.cpp:316, at /home/jenkins-agent/workspace/nightly_linux_repeated/ws/src/ros2/rcl/rcl/src/rcl/node.c:261
```

ivanpauno, 2020-11-03:
> [D0108|ENABLE]DDS_DomainParticipant_enableI:Automatic participant index failed to initialize. PLEASE VERIFY CONSISTENT TRANSPORT / DISCOVERY CONFIGURATION. 
DDSDomainParticipant_impl::createI:ERROR: Failed to auto-enable entity

This sounds like a random `rti Connext` bug, I'm not sure we can do something about it.

chapulina, 2020-11-03:
The test passed in the latest build.

I think we should keep an eye on it. If it turns out to be an spurious failure, I think we can close this issue. But if it turns out to be flaky and out of our control, I'd suggest disabling the test for Connext until it can be fixed.

ivanpauno, 2020-11-30:
The test hasn't failed in the last 26 nightlies.

Based on the post [here](https://community.rti.com/forum-topic/please-verify-constistent-transport-discovery-configuration), this is a super random failure when `Connext` fails to find a network interface (one of the posts is from a ROS 2 developer reporting having seen this error in nightlies :wink:).

Thought it would be great to solve this, it seems to be super flaky (I don't remember to have seen it before) and it doesn't seem to be something we can fix without rti help, so I will close for the moment.

@chapulina feel free to reopen if you think this is still valid.

closed 2020-11-30 by ivanpauno