ros2/rcl #856: 👩‍🌾 test_info_by_topic flaky on Linux

chapulina, 2020-11-04:
The test is flaky for CycloneDDS on Linux aarch64 (failed 6/24 latest runs):

https://ci.ros2.org/view/nightly/job/nightly_linux-aarch64_repeated/1398/testReport/projectroot/test/test_info_by_topic__rmw_cyclonedds_cpp/history/

It's also slightly flaky for FastRTPS on aarch64 (failed 1/24 latest runs):

https://ci.ros2.org/view/nightly/job/nightly_linux-aarch64_repeated/1398/testReport/projectroot/test/test_info_by_topic__rmw_fastrtps_cpp/history/

On x86_64, it passes consistently for CycloneDDS:

https://ci.ros2.org/view/nightly/job/nightly_linux_repeated/2096/testReport/projectroot/test/test_info_by_topic__rmw_cyclonedds_cpp/history/

But it's slightly flaky for FastRTPS (failed 1/24 latest runs): 

https://ci.ros2.org/view/nightly/job/nightly_linux_repeated/2096/testReport/projectroot/test/test_info_by_topic__rmw_fastrtps_cpp/history/

Failure log for the flakiest case:

```
[ RUN      ] TestInfoByTopicFixture__rmw_cyclonedds_cpp.test_rcl_get_publishers_subscription_info_by_topic
/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/src/ros2/rcl/rcl/test/rcl/test_info_by_topic.cpp:367: Failure
Expected equality of these values:
  topic_endpoint_info_array_pub.size
    Which is: 0
  1u
    Which is: 1
Expected one publisher
[  FAILED  ] TestInfoByTopicFixture__rmw_cyclonedds_cpp.test_rcl_get_publishers_subscription_info_by_topic (16 ms)
```

fujitatomoya, 2020-11-05:
adding extra sleep in [here](https://github.com/ros2/rcl/blob/8a495c0e35a3352d64f0aa5fa5fa46feb326b483/rcl/test/rcl/test_info_by_topic.cpp#L359) would be okay enough, or maybe we can call `wait_for_subscription_to_be_ready` instead.

fujitatomoya, 2020-11-05:
https://github.com/ros2/rcl/pull/859 should fix the flaky situation. I tried more than 30 times but i was unable to reproduce this issue.

fujitatomoya, 2020-11-05:
@chapulina 
CC: @clalancette 

friendly ping.

fujitatomoya, 2020-11-10:
@chapulina 
CC: @clalancette 

i think we can close this one, since https://github.com/ros2/rcl/pull/859 is merged.

chapulina, 2020-11-10:
Thanks. Closing for now, please reopen if the test starts flaking again.

closed 2020-11-10 by chapulina