ros2/rcl #861: 👨‍🌾 TestNodeFixture__rmw_fastrtps_cpp.test_rcl_node_init_with_internal_errors fails on nightly OSX repeated

azeey, 2020-11-17:
This has been failing since https://ci.ros2.org/view/nightly/job/nightly_osx_repeated/2103/testReport/junit/projectroot/test/test_node__rmw_fastrtps_cpp/. Since then it has failed 22 out of the last 30 tests.

<details>

Stacktrace:

```
-- run_test.py: extra environment variables:
 - DYLD_INSERT_LIBRARIES=/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/install/osrf_testing_tools_cpp/lib/libmemory_tools_interpose.dylib
 - RMW_IMPLEMENTATION=rmw_fastrtps_cpp
-- run_test.py: extra environment variables to append:
 - DYLD_LIBRARY_PATH+=/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/build/rcl
-- run_test.py: invoking following command in '/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/build/rcl/test':
 - /Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/build/rcl/test/test_node__rmw_fastrtps_cpp --gtest_output=xml:/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/build/rcl/test_results/rcl/test_node__rmw_fastrtps_cpp.gtest.xml
Running main() from /Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/install/gtest_vendor/src/gtest_vendor/src/gtest_main.cc
[==========] Running 9 tests from 1 test suite.
[----------] Global test environment set-up.
[----------] 9 tests from TestNodeFixture__rmw_fastrtps_cpp
[ RUN      ] TestNodeFixture__rmw_fastrtps_cpp.test_rcl_node_accessors
[       OK ] TestNodeFixture__rmw_fastrtps_cpp.test_rcl_node_accessors (65 ms)
[ RUN      ] TestNodeFixture__rmw_fastrtps_cpp.test_rcl_node_life_cycle
[       OK ] TestNodeFixture__rmw_fastrtps_cpp.test_rcl_node_life_cycle (16 ms)
[ RUN      ] TestNodeFixture__rmw_fastrtps_cpp.test_rcl_node_init_with_internal_errors

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'error not set, at /Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rcl/rcl/src/rcl/node.c:262'

with this new error message:

  'rcl node's rmw handle is invalid, at /Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rcl/rcl/src/rcl/node.c:414'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [1605438092.894300311] [rcl]: Failed to fini publisher for node: 1

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'failed to format node namespace string, at /Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rcl/rcl/src/rcl/node.c:181'

with this new error message:

  'rcl node implementation is invalid, at /Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rcl/rcl/src/rcl/node.c:412'

rcutils_reset_error() should be called after error handling to avoid this.
<<<

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'failed to format node namespace string, at /Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rcl/rcl/src/rcl/node.c:181'

with this new error message:

  'rcl node implementation is invalid, at /Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rcl/rcl/src/rcl/node.c:412'

rcutils_reset_error() should be called after error handling to avoid this.
<<<

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'failed to format node namespace string, at /Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rcl/rcl/src/rcl/node.c:181'

with this new error message:

  'rcl node implementation is invalid, at /Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rcl/rcl/src/rcl/node.c:412'

[... 485 lines cut ...]
/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rcutils/src/snprintf.c:42 Injecting fault and executing {(*__error()) = 22; return -1;}
/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rcutils/src/allocator.c:34 Injecting fault and returning ((void*)0)
/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rcutils/src/snprintf.c:42 Injecting fault and executing {(*__error()) = 22; return -1;}
/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rcutils/src/allocator.c:34 Injecting fault and returning ((void*)0)
/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rcutils/src/snprintf.c:42 Injecting fault and executing {(*__error()) = 22; return -1;}
/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rcutils/src/allocator.c:34 Injecting fault and returning ((void*)0)
/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rcutils/src/snprintf.c:42 Injecting fault and executing {(*__error()) = 22; return -1;}
/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/participant.cpp:149 Injecting fault and returning nullptr
/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rcutils/src/get_env.c:32 Injecting fault and returning "some string error"
/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rcutils/src/get_env.c:32 Injecting fault and returning "some string error"
/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp/src/publisher.cpp:57 Injecting fault and returning nullptr
/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rcutils/src/allocator.c:34 Injecting fault and returning ((void*)0)
/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rcutils/src/allocator.c:34 Injecting fault and returning ((void*)0)
/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp/src/subscription.cpp:62 Injecting fault and returning nullptr
/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rcutils/src/allocator.c:34 Injecting fault and returning ((void*)0)
/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rcutils/src/strdup.c:32 Injecting fault and returning ((void*)0)
/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rcutils/src/strdup.c:43 Injecting fault and returning ((void*)0)
/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rcutils/src/allocator.c:34 Injecting fault and returning ((void*)0)
/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_guard_condition.cpp:27 Injecting fault and returning nullptr
/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/listener_thread.cpp:47 Injecting fault and returning 1
-- run_test.py: return code -6
-- run_test.py: generate result file '/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/build/rcl/test_results/rcl/test_node__rmw_fastrtps_cpp.gtest.xml' with failed test
-- run_test.py: verify result file '/Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/build/rcl/test_results/rcl/test_node__rmw_fastrtps_cpp.gtest.xml'
```
</details>

ivanpauno, 2020-12-14:
Based on the error output:

> /Users/osrf/jenkins-agent/workspace/nightly_osx_repeated/ws/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/listener_thread.cpp:47 Injecting fault and returning 1
-- run_test.py: return code -6

I assume that we're not handling correctly when [this function](https://github.com/ros2/rmw_fastrtps/blob/867514bc2e529c97e7b779d112d9eab0ff96e615/rmw_fastrtps_shared_cpp/src/listener_thread.cpp#L47) fails, though I will have to dig a bit deeper to understand the problem.

This only seems to be happening on macOS

clalancette, 2022-02-03:
Closing as macOS is no longer being run nightly.

closed 2022-02-03 by clalancette