ros2/rcl #869: failed to install rcl_yaml_param_parser package

mkeyno, 2020-11-28:
I've tried to install ROS2 foxy from source to my ubuntu Linux, but at will stop to compile sources at the `rcl_yaml_param_parser package`

I've checked the installed libraries and find the `python3-yaml ` and 'libyaml-dev` already  installed 
```
Starting >>> rcl_yaml_param_parser
--- stderr: rcl_yaml_param_parser
CMake Error at CMakeLists.txt:7 (find_package):
  By not providing "Findlibyaml_vendor.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "libyaml_vendor", but CMake did not find one.

  Could not find a package configuration file provided by "libyaml_vendor"
  with any of the following names:

    libyaml_vendorConfig.cmake
    libyaml_vendor-config.cmake

  Add the installation prefix of "libyaml_vendor" to CMAKE_PREFIX_PATH or set
  "libyaml_vendor_DIR" to a directory containing one of the above files.  If
  "libyaml_vendor" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< rcl_yaml_param_parser [1.43s, exited with code 1]
```
and finally, it affects rcl package
```
Starting >>> rcl
--- stderr: rcl
CMake Error at CMakeLists.txt:8 (find_package):
  By not providing "Findrcl_yaml_param_parser.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "rcl_yaml_param_parser", but CMake did not find one.

  Could not find a package configuration file provided by
  "rcl_yaml_param_parser" with any of the following names:

    rcl_yaml_param_parserConfig.cmake
    rcl_yaml_param_parser-config.cmake

  Add the installation prefix of "rcl_yaml_param_parser" to CMAKE_PREFIX_PATH
  or set "rcl_yaml_param_parser_DIR" to a directory containing one of the
  above files.  If "rcl_yaml_param_parser" provides a separate development
  package or SDK, be sure it has been installed.

---
Failed   <<< rcl [5.71s, exited with code 1]

```

fujitatomoya, 2020-11-29:
@mkeyno 

can you check that libyaml_vendor repository is in your colcon workspace? if you use `vcs import`, it should be under `<colcon_ws>/src/ros2/libyaml_vendor`.

https://github.com/ros2/ros2/blob/foxy/ros2.repos#L218-L221

mkeyno, 2020-11-30:
@fujitatomoya 
thanks for the response, I install the last update of ROS2 from source and use `vcs import`

```
ros2/libyaml_vendor:
    type: git
    url: https://github.com/ros2/libyaml_vendor.git
    version: foxy
```
I think CMake does not configure correctly to find the installed library in the system

fujitatomoya, 2020-11-30:
did you follow procedure https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Development-Setup/ ? i don't see any problems...

mkeyno, 2020-12-04:
@fujitatomoya 
Of course, but I think to solve this issue, it should manually localize the installed library to the script
is there a way to log how the script searches for the library and what result it finds?

fujitatomoya, 2020-12-05:
> is there a way to log how the script searches for the library and what result it finds?

sorry, i do not know how to do that either. if i find that out, i will let you know.

ivanpauno, 2020-12-21:
is there a `<your_ws>/install/libyaml_vendor` folder? Check that after running `colcon build`, and the build of `rcl_yaml_param_parser` fails (if you used `--merge-install`, you should be able to find a `<your_ws>/install/share/libyaml_vendor` folder)

can you copy exactly the steps you used to build the workspace?

ivanpauno, 2020-12-21:
> 'libyaml-dev` already installed

it would also be interesting if you can try again after removing `libyaml-dev` (no need to remove `python3-yaml`).

mkeyno, 2020-12-29:
@ivanpauno thanks for the reply
yes there is a couple of folders and I used --merge-install for build 
[image]
[image]

ivanpauno, 2021-01-04:
> it would also be interesting if you can try again after removing libyaml-dev (no need to remove python3-yaml).

@mkeyno did you try this?
I mean doing `sudo apt purge libyaml-dev` and building your workspace again.
`yaml_vendor` is supposed to use your system installed `libyaml` if available, and if not it will build it from src.
AFAIR, we always test with libyaml built from src (and our install steps usually don't install system libyaml), so this is probably a problem in the cmake logic in `yaml_vendor`.

PS: If you cannot uninstall `libyaml-dev` from your system because you're using it for something else, please try without/with it in a docker environment to confirm this is the issue.
PS2: Prefer copying text and not images images, e.g. you can copy the text output of colcon to a code block.