ros2/rcl #870: TestSubscriptionFixture__rmw_connext_cpp.test_subscription_nominal_string_sequence fails on nightly Linux repeated

jacobperron, 2020-11-30:
The test has been failing since November 28th: https://ci.ros2.org/job/nightly_linux_repeated/2120/testReport/rcl/TestSubscriptionFixture__rmw_connext_cpp/test_subscription_nominal_string_sequence/

```
/home/jenkins-agent/workspace/nightly_linux_repeated/ws/src/ros2/rcl/rcl/test/rcl/test_subscription.cpp:484
Expected equality of these values:
  3u
    Which is: 3
  messages.size
    Which is: 1
```

ivanpauno, 2020-12-21:
The issue is that the test is first [publishing 3 messages](https://github.com/ros2/rcl/blob/2ee91d8e4c02ce04899c792c1b28119673b5bf61/rcl/test/rcl/test_subscription.cpp#L422-L426) and "immediately" expecting them to be [available in a subscription](https://github.com/ros2/rcl/blob/2ee91d8e4c02ce04899c792c1b28119673b5bf61/rcl/test/rcl/test_subscription.cpp#L482-L485).

Maybe, the test should call `rcl_take_sequence` in a loop with a timeout expecting that the total number of messages received is 3, that would make the test more reliable.
Particularly, this seems to only be failing with connext, which is usually the one that needs more "time" to avoid these issues.

@jacobperron Does that sound good to you ?

jacobperron, 2020-12-21:
hmm, but isn't the purpose of this line to make sure that messages are ready before taking them?

https://github.com/ros2/rcl/blob/2ee91d8e4c02ce04899c792c1b28119673b5bf61/rcl/test/rcl/test_subscription.cpp#L430

ivanpauno, 2020-12-22:
> hmm, but isn't the purpose of this line to make sure that messages are ready before taking them?

That ensures that the subscription has at least one message ready, but it doesn't mean that it received the other two.
I don't think that we have a way of waiting until the three messages are ready (we cannot check how many messages a subscription has in its history queue).

jacobperron, 2020-12-22:
> That ensures that the subscription has at least one message ready, but it doesn't mean that it received the other two.

I see. Yeah, then I suppose adding a loop with a timeout makes sense. Maybe also adding a comment above the `wait_for_subscription_to_be_ready` statement, describing the limitation.

closed 2020-12-23 by ivanpauno