ros2/rcl #876: Failed to get trigger guard condition in jump callback

ThomasRFischer, 2020-12-16:
## Bug report

**Required Info:**

- Operating System:
  - 
Ubuntu 18.04.5 LTS
- Installation type:
binaries 
- Version or commit hash:
  ROS2 FOXY Development
- DDS implementation:
Cyclone DDS Version 0.7.0 https://github.com/ros2/rmw_cyclonedds
- RMW
ROS 2 RMW for Eclipse Cyclone DDS Version 2.2.0 https://github.com/ros2/rmw
 
- Client library (if applicable):
  - 

#### Steps to reproduce issue
The issue is intermittent 

#### Expected behavior
It should run without error 

#### Actual behavior
[planner_server-21] [ERROR 1605305996.010625274] [rcl]: Failed to get trigger guard condition in jump callback (/ros2_foxy_build/core/src/ros2/rcl/rcl/src/rcl/timer.c:114)
[planner_server-21]
[planner_server-21] >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
[planner_server-21] This error state is being overwritten:
[planner_server-21]
[planner_server-21]   'guard_condition_handle not from this implementation, at /ros2_foxy_build/core/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/rmw_node.cpp:2713, at /ros2_foxy_build/core/src/ros2/rcl/rcl/src/rcl/guard_condition.c:160'
[planner_server-21]
[planner_server-21] with this new error message:
[planner_server-21]
[planner_server-21]   'guard condition implementation is invalid, at /ros2_foxy_build/core/src/ros2/rcl/rcl/src/rcl/guard_condition.c:174'
[planner_server-21]
[planner_server-21] rcutils_reset_error() should be called after error handling to avoid this.
[planner_server-21] <<<
[planner_server-21] [ERROR 1605306298.359010300] [rcl]: Failed to get trigger guard condition in jump callback (/ros2_foxy_build/core/src/ros2/rcl/rcl/src/rcl/timer.c:114)
[ERROR] [planner_server-21]: process has died [pid 12676, exit code -11, cmd '/opt/ros/foxy/nav2_planner/lib/nav2_planner/planner_server --ros-args -r __node:=planner_server --params-file /tmp/tmpp28qz1d4 -r /tf:=tf -r /tf_static:=tf_static'].

#### Additional information
ROS2 Costmap Planner application

clalancette, 2020-12-17:
@ThomasRFischer Is there any chance you could give the code in #877 a whirl?  It is obviously a bug, so we'll go forward with it regardless, but additional confirmation that it fixes your problem would be welcome.  Thanks.

ThomasRFischer, 2020-12-18:
Thanks for looking into this. I will try to get back to testing in next week, Monday or Tuesday. I will reply to the results of the test in ticket #877.

PR #877 (2021-01-07): Move the guard condition cleanup after removing callback.
Otherwise, the callback could attempt to take a reference
to a guard condition that has already been deleted.

This should fix #876 , but this is still a draft since I'm doing additional testing on it.
closed 2021-01-07 by clalancette