ros2/rcl #886: Program termination with Ctrl+C stops in the middle.

hsgwa, 2021-01-26:
## Bug report

- Operating System:
  - Ubuntu 20.04
- Installation type:
  -  from source
- Version or commit hash:
  -  `2883ce4` (master)
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp or rclpy

#### Steps to reproduce issue
This issue occurred in both rclcpp and rclpy.
```
$ ros2 run demo_nodes_cpp talker --ros-args --log-level debug
# send Ctrl+C
```
```
$ ros2 run demo_nodes_py talker --ros-args --log-level debug
# send Ctrl+C
```

#### Expected behavior
terminate safely.

#### Actual behavior
stop during finalizing state.
It seems like node finalizing has some problem.

```
$ ros2 run demo_nodes_cpp talker --ros-args --log-level debug
^C[INFO] [1611636125.400290516] [rclcpp]: signal_handler(signal_value=2)
[DEBUG] [1611636125.400375072] [rclcpp]: signal_handler(): SIGINT received, notifying deferred signal handler
[DEBUG] [1611636125.400436017] [rclcpp]: deferred_signal_handler(): woken up due to SIGINT or uninstall
[DEBUG] [1611636125.400486808] [rclcpp]: deferred_signal_handler(): SIGINT received, shutting down
[DEBUG] [1611636125.400651593] [rclcpp]: deferred_signal_handler(): shutting down rclcpp::Context @ 0x55d466e8e830, because it had shutdown_on_sigint == true
[DEBUG] [1611636125.400687545] [rcl]: Shutting down ROS client library, for context at address: 0x55d466e8eab0
[DEBUG] [1611636125.400782572] [rcl]: Guard condition in wait set is ready
[DEBUG] [1611636125.400830210] [rcl]: Finalizing publisher
[DEBUG] [1611636125.401084234] [rcl]: Timer canceled
[DEBUG] [1611636125.401109610] [rcl]: Timer canceled
[DEBUG] [1611636125.401343531] [rcl]: Publisher finalized
[DEBUG] [1611636125.401495158] [rclcpp]: deferred_signal_handler(): waiting for SIGINT or uninstall
[DEBUG] [1611636125.401520824] [rcl]: Finalizing event
[DEBUG] [1611636125.401541437] [rcl]: Event finalized
[DEBUG] [1611636125.401579052] [rcl]: Finalizing publisher
[DEBUG] [1611636125.401832104] [rcl]: Publisher finalized
[DEBUG] [1611636125.402000585] [rcl]: Finalizing event
[DEBUG] [1611636125.402013763] [rcl]: Event finalized
[DEBUG] [1611636125.402026546] [rcl]: Finalizing subscription
[DEBUG] [1611636125.402194399] [rcl]: Subscription finalized
[DEBUG] [1611636125.402357121] [rcl]: Finalizing service
[DEBUG] [1611636125.404965197] [rcl]: Service finalized
[DEBUG] [1611636125.405047754] [rcl]: Finalizing service
[DEBUG] [1611636125.405435803] [rcl]: Service finalized
[DEBUG] [1611636125.405483505] [rcl]: Finalizing service
[DEBUG] [1611636125.405815296] [rcl]: Service finalized
[DEBUG] [1611636125.405853974] [rcl]: Finalizing service
[DEBUG] [1611636125.406083684] [rcl]: Service finalized
[DEBUG] [1611636125.406120386] [rcl]: Finalizing service
[DEBUG] [1611636125.406316801] [rcl]: Service finalized
[DEBUG] [1611636125.406351774] [rcl]: Finalizing service
[DEBUG] [1611636125.406559456] [rcl]: Service finalized
[DEBUG] [1611636125.406635210] [rcl]: Finalizing event
[DEBUG] [1611636125.406648754] [rcl]: Event finalized
[DEBUG] [1611636125.406664502] [rcl]: Finalizing publisher
[DEBUG] [1611636125.406853173] [rcl]: Publisher finalized
[DEBUG] [1611636125.407209588] [rcl]: Finalizing node
```

```
$ ros2 run demo_nodes_py talker --ros-args --log-level debug
^C[DEBUG] [1611636186.002509447] [rcl]: Guard condition in wait set is ready
[DEBUG] [1611636186.002854722] [rcl]: Finalizing event
[DEBUG] [1611636186.002893463] [rcl]: Event finalized
[DEBUG] [1611636186.002922664] [rcl]: Finalizing publisher
[DEBUG] [1611636186.003449009] [rcl]: Publisher finalized
[DEBUG] [1611636186.003530384] [rcl]: Finalizing event
[DEBUG] [1611636186.003552872] [rcl]: Event finalized
[DEBUG] [1611636186.003578862] [rcl]: Finalizing publisher
[DEBUG] [1611636186.004997261] [rcl]: Publisher finalized
[DEBUG] [1611636186.005102081] [rcl]: Finalizing service
[DEBUG] [1611636186.006543735] [rcl]: Service finalized
[DEBUG] [1611636186.006641986] [rcl]: Finalizing service
[DEBUG] [1611636186.006956990] [rcl]: Service finalized
[DEBUG] [1611636186.007011764] [rcl]: Finalizing service
[DEBUG] [1611636186.008383436] [rcl]: Service finalized
[DEBUG] [1611636186.008476722] [rcl]: Finalizing service
[DEBUG] [1611636186.009434476] [rcl]: Service finalized
[DEBUG] [1611636186.009525846] [rcl]: Finalizing service
[DEBUG] [1611636186.010021857] [rcl]: Service finalized
[DEBUG] [1611636186.010073573] [rcl]: Finalizing service
[DEBUG] [1611636186.011492397] [rcl]: Service finalized
[DEBUG] [1611636186.011578826] [rcl]: Timer canceled
[DEBUG] [1611636186.011635771] [rcl]: Finalizing node
[DEBUG] [1611636186.011656284] [rcl]: Finalizing publisher
[DEBUG] [1611636186.011833613] [rcl]: Publisher finalized
```

hsgwa, 2021-01-27:
The locked state is as follows.

```
(gdb) info thread
       Id   Target Id                                        Frame
 1    Thread 0x7ff21f778000 (LWP 751366) "talker"      __pthread_clockjoin_ex (threadid=140677602068224,
    thread_return=0x7fff0d029590, clockid=<optimized out>, abstime=<optimized out>, block=<optimized out>)
    at pthread_join_common.c:145
  2    Thread 0x7ff21f771700 (LWP 751367) "talker-ust"  __libc_recvmsg (flags=0, msg=0x7ff21f76f840, fd=3)
    at ../sysdeps/unix/sysv/linux/recvmsg.c:28
  3    Thread 0x7ff21ef70700 (LWP 751368) "talker-ust"  syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  4    Thread 0x7ff21e078700 (LWP 751369) "talker"      futex_abstimed_wait_cancelable (private=0, abstime=0x0, clockid=0,
    expected=0, futex_word=0x7ff2208dd9e0 <rclcpp::SignalHandler::signal_handler_sem_>) at ../sysdeps/nptl/futex-internal.h:320
  5    Thread 0x7ff21d4d8700 (LWP 751370) "gc"          futex_abstimed_wait_cancelable (private=<optimized out>,
    abstime=0x7ff21d4d74b0, clockid=<optimized out>, expected=0, futex_word=0x55957781eb74) at ../sysdeps/nptl/futex-internal.h:320
  6    Thread 0x7ff21ccd7700 (LWP 751371) "dq.builtins" futex_wait_cancelable (private=<optimized out>, expected=0,
    futex_word=0x559577817ed0) at ../sysdeps/nptl/futex-internal.h:183
  7    Thread 0x7ff20ffff700 (LWP 751372) "dq.user"     futex_wait_cancelable (private=<optimized out>, expected=0,
    futex_word=0x559577824540) at ../sysdeps/nptl/futex-internal.h:183
  8    Thread 0x7ff20f7fe700 (LWP 751373) "tev"         futex_abstimed_wait_cancelable (private=<optimized out>,
    abstime=0x7ff20f7fd4d0, clockid=<optimized out>, expected=0, futex_word=0x55957781ea78) at ../sysdeps/nptl/futex-internal.h:320
  9    Thread 0x7ff20e7fc700 (LWP 751375) "recvMC"      __libc_recvmsg (flags=0, msg=0x7ff20e7fb1b0, fd=8)
    at ../sysdeps/unix/sysv/linux/recvmsg.c:28
```
```
(gdb) thread 1
(gdb) bt
#0  __pthread_clockjoin_ex (threadid=140677602068224, thread_return=0x7fff0d029590, clockid=<optimized out>, abstime=<optimized out>,
    block=<optimized out>) at pthread_join_common.c:145
#1  0x00007ff21e588f44 in ddsrt_thread_join (thread=..., thread_result=0x0)
    at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/threads/posix/threads.c:380
#2  0x00007ff21e53c23f in join_thread (ts1=0x559577810480)
    at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/q_thread.c:381
#3  0x00007ff21e51f40b in wait_for_receive_threads (gv=0x559577818a30)
    at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/q_init.c:795
#4  0x00007ff21e522bfc in rtps_stop (gv=0x559577818a30)
    at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/q_init.c:1761
#5  0x00007ff21e55b884 in dds_domain_free (vdomain=0x559577818790)
    at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/core/ddsc/src/dds_domain.c:311
#6  0x00007ff21e55ffb3 in dds_entity_deriver_delete (e=0x559577818790)
    at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/core/ddsc/src/dds__types.h:193
#7  0x00007ff21e560ec8 in really_delete_pinned_closed_locked (e=0x559577818790, delstate=DIS_EXPLICIT)
    at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/core/ddsc/src/dds_entity.c:522
#8  0x00007ff21e560bab in dds_delete_impl_pinned (e=0x559577818790, delstate=DIS_EXPLICIT)
    at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/core/ddsc/src/dds_entity.c:409
#9  0x00007ff21e560b20 in dds_delete_impl (entity=1934017353, delstate=DIS_EXPLICIT)
    at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/core/ddsc/src/dds_entity.c:384
#10 0x00007ff21e560a60 in dds_delete (entity=1934017353)
    at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/core/ddsc/src/dds_entity.c:367
#11 0x00007ff21e68e7cd in check_destroy_domain (domain_id=0)
    at /home/hasegawa/ros2_rolling/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/rmw_node.cpp:826
#12 0x00007ff21e69012b in rmw_context_impl_t::clean_up (this=0x55957780f1c0)
    at /home/hasegawa/ros2_rolling/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/rmw_node.cpp:1129
#13 0x00007ff21e6901aa in rmw_context_impl_t::fini (this=0x55957780f1c0)
    at /home/hasegawa/ros2_rolling/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/rmw_node.cpp:1140
#14 0x00007ff21e691d27 in rmw_destroy_node (node=0x5595778380d0)
    at /home/hasegawa/ros2_rolling/src/ros2/rmw_cyclonedds/rmw_cyclonedds_cpp/src/rmw_node.cpp:1372
#15 0x00007ff21fb343ec in rmw_destroy_node (v1=0x5595778380d0)
    at /home/hasegawa/ros2_rolling/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:220
#16 0x00007ff21fb9a7f3 in rcl_node_fini (node=0x5595778186c0) at /home/hasegawa/ros2_rolling/src/ros2/rcl/rcl/src/rcl/node.c:382
#17 0x00007ff220684bf8 in rclcpp::node_interfaces::NodeBase::<lambda(rcl_node_t*)>::operator()(rcl_node_t *) const (
    __closure=0x55957783d120, node=0x5595778186c0)
    at /home/hasegawa/ros2_rolling/src/ros2/rclcpp/rclcpp/src/rclcpp/node_interfaces/node_base.cpp:141
#18 0x00007ff22068697e in std::_Sp_counted_deleter<rcl_node_t*, rclcpp::node_interfaces::NodeBase::NodeBase(const string&, const string&, rclcpp::Context::SharedPtr, const rcl_node_options_t&, bool, bool)::<lambda(rcl_node_t*)>, std::allocator<void>, (__gnu_cxx::_Lock_policy)2>::_M_dispose(void) (this=0x55957783d110) at /usr/include/c++/9/bits/shared_pt [... cut ...]
#44 0x0000559575933bc4 in std::vector<rclcpp_components::NodeInstanceWrapper, std::allocator<rclcpp_components::NodeInstanceWrapper> >::_M_erase_at_end (this=0x7fff0d02c7d0, __pos=0x559577879270) at /usr/include/c++/9/bits/stl_vector.h:1793
#45 0x0000559575932432 in std::vector<rclcpp_components::NodeInstanceWrapper, std::allocator<rclcpp_components::NodeInstanceWrapper> >::clear (this=0x7fff0d02c7d0) at /usr/include/c++/9/bits/stl_vector.h:1496
#46 0x00005595759302d3 in main (argc=1, argv=0x7fff0d02d648)
    at /home/hasegawa/ros2_rolling/build/demo_nodes_cpp/rclcpp_components/node_main_talker.cpp:61
```

fujitatomoya, 2021-01-27:
@ivanpauno 

could you help to move this into [rmw_cyclonedds](https://github.com/ros2/rmw_cyclonedds) since this only happens on cyclonedds.

ivanpauno, 2021-01-27:
@hsgwa can you share a multhithreaded backtrace (`thread apply all bt`)?

ivanpauno, 2021-01-27:
> could you help to move this into rmw_cyclonedds since this only happens on cyclonedds.

It happens only with `rmw_cyclonedds`, but it's not clear yet if that's a bug in cyclonedds or some ros 2 package.
I will keep the bug report here until more information is collected

hsgwa, 2021-01-28:
@ivanpauno 
>  @hsgwa can you share a multhithreaded backtrace (thread apply all bt)?

here it is.
I added version info of related packages to the original post.

```
(gdb) thread apply all bt
Thread 9 (Thread 0x7fcf2b7fe700 (LWP 3519)):
#0  __libc_recvmsg (flags=0, msg=0x7fcf2b7fd1b0, fd=8) at ../sysdeps/unix/sysv/linux/recvmsg.c:28
#1  __libc_recvmsg (fd=8, msg=0x7fcf2b7fd1b0, flags=0) at ../sysdeps/unix/sysv/linux/recvmsg.c:25
#2  0x00007fcf432b52fc in ddsrt_recvmsg (sock=8, msg=0x7fcf2b7fd1b0, flags=0, rcvd=0x7fcf2b7fd188) at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/sockets/posix/socket.c:455
#3  0x00007fcf431e7c10 in ddsi_udp_conn_read (conn_cmn=0x55a37be0a7c0, buf=0x7fcf40000070 "", len=65536, allow_spurious=true, srcloc=0x7fcf2b7fd430) at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/ddsi_udp.c:90
#4  0x00007fcf431e6be5 in ddsi_conn_read (conn=0x55a37be0a7c0, buf=0x7fcf40000070 "", len=65536, allow_spurious=true, srcloc=0x7fcf2b7fd430) at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/security/api/../../core/ddsi/include/dds/ddsi/ddsi_tran.h:254
#5  0x00007fcf4326728c in do_packet (ts1=0x55a37be02500, gv=0x55a37be0afa0, conn=0x55a37be0a7c0, guidprefix=0x0, rbpool=0x55a37be17f50) at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/q_receive.c:3252
#6  0x00007fcf43268172 in recv_thread (vrecv_thread_arg=0x55a37be0dde0) at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/q_receive.c:3500
#7  0x00007fcf43269c88 in create_thread_wrapper (ptr=0x55a37be02500) at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/q_thread.c:241
#8  0x00007fcf432b67aa in os_startRoutineWrapper (threadContext=0x55a37be17fb0) at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/threads/posix/threads.c:165
#9  0x00007fcf446c7609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#10 0x00007fcf44a23293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 8 (Thread 0x7fcf40a02700 (LWP 3517)):
#0  futex_abstimed_wait_cancelable (private=<optimized out>, abstime=0x7fcf40a014d0, clockid=<optimized out>, expected=0, futex_word=0x55a37be0a498) at ../sysdeps/nptl/futex-internal.h:320
#1  __pthread_cond_wait_common (abstime=0x7fcf40a014d0, clockid=<optimized out>, mutex=0x55a37be0a448, cond=0x55a37be0a470) at pthread_cond_wait.c:520
#2  __pthread_cond_timedwait (cond=0x55a37be0a470, mutex=0x55a37be0a448, abstime=0x7fcf40a014d0) at pthread_cond_wait.c:656
#3  0x00007fcf432b60fb in ddsrt_cond_waituntil (cond=0x55a37be0a470, mutex=0x55a37be0a448, abstime=1611795496341632475) at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/sync/posix/sync.c:116
#4  0x00007fcf432b61bd in ddsrt_cond_waitfor (cond=0x55a37be0a470, mutex=0x55a37be0a448, reltime=999899872) at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/sync/posix/sync.c:137
#5  0x00007fcf432726a1 in xevent_thread (xevq=0x55a37be0a3f0) at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/q_xevent.c:1286
#6  0x00007fcf43269c88 in create_thread_wrapper (ptr=0x55a37be02480) at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/q_thread.c:241
#7  0x00007fcf432b67aa in os_startRoutineWrapper (threadContext=0x55a37be112a0) at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/threads/posix/threads.c:165
#8  0x00007fcf446c7609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#9  0x00007fcf44a23293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 7 (Thread 0x7fcf41203700 (LWP 3516)):
#0  futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0x55a37be0a2d0) at ../sysdeps/nptl/futex-internal.h:183
#1  __pthread_cond_wait_common (abstime=0x0, clockid=0, mutex=0x55a37be0a280, cond=0x55a37be0a2a8) at pthread_cond_wait.c:508
#2  __pthread_cond_wait (cond=0x55a37be0a2a8, mutex=0x55a37be0a280) at pthread_cond_wait.c:638
#3  0x00007fcf432b5fc2 in ddsrt_cond_wait (cond=0x55a37be0a2a8, mutex=0x55a37be0a280) at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/sync/posix/sync.c:92
#4  0x00007fcf4325ab88 in dqueue_thread (q=0x55a37be0a280) at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/q_radmin.c:2502
#5  0x00007fcf43269c88 in create_thread_wrapper (ptr=0x55a37be02440) at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/q_thread.c:241
#6  0x00007fcf432b67aa in os_startRoutineWrapper (threadContext=0x55a37be110d0) at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/threads/posix/threads.c:165
#7  0x00007fcf446c7609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#8  0x00007fcf44a23293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 6 (Thread 0x7fcf41a04700 (LWP 3515)):
#0  futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0x55a37be0a230) at ../sysdeps/nptl/futex-internal.h:183
#1  __pthread_cond_wait_common (abstime=0x0, clockid=0, mutex=0x55a37be0a1e0, cond=0x55a37be0a208) at pthread_cond_wait.c:508
#2  __pthread_cond_wait (cond=0x55a37be0a208, mutex=0x55a37be0a1e0) at pthread_cond_wait.c:638
[... cut ...]
#44 0x000055a37a90bbc4 in std::vector<rclcpp_components::NodeInstanceWrapper, std::allocator<rclcpp_components::NodeInstanceWrapper> >::_M_erase_at_end (this=0x7ffd03863220, __pos=0x55a37be6af50) at /usr/include/c++/9/bits/stl_vector.h:1793
#45 0x000055a37a90a432 in std::vector<rclcpp_components::NodeInstanceWrapper, std::allocator<rclcpp_components::NodeInstanceWrapper> >::clear (this=0x7ffd03863220) at /usr/include/c++/9/bits/stl_vector.h:1496
#46 0x000055a37a9082d3 in main (argc=4, argv=0x7ffd03864098) at /home/hasegawa/ros2_rolling/build/demo_nodes_cpp/rclcpp_components/node_main_talker.cpp:61
```

ivanpauno, 2021-01-28:
I wasn't able to reproduce the issue, the process always exited correctly at the first `ctrl-c`.

@hsgwa does pressing 2 or 3 times ctrl-c work for you?
can you double check that the steps provided to reproduce the issue are correct?
Thanks!

hsgwa, 2021-01-29:
@ivanpauno 

Thanks for checking!
I haven't established steps to reproduce yet though, I share what I checked.

1. This process doesn't terminate if several times ctrl-C.
```
$ ros2 run demo_nodes_cpp talker --ros-args --log-level debug
[DEBUG] [1611907728.513152992] [rclcpp]: signal handler installed
[DEBUG] [1611907728.513237716] [rclcpp]: deferred_signal_handler(): waiting for SIGINT or uninstall
[DEBUG] [1611907728.528554216] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '0' services
[DEBUG] [1611907728.528733222] [talker]: Load library libtopics_library.so
[DEBUG] [1611907728.783527729] [talker]: Instantiate class rclcpp_components::NodeFactoryTemplate<demo_nodes_cpp::Talker>
[DEBUG] [1611907728.783841080] [rcl]: Couldn't parse arg 0 (/home/hasegawa/ros2_rolling/install/demo_nodes_cpp/lib/demo_nodes_cpp/talker) as a remap rule in its deprecated form. Error: Expected lexeme type (19) not found, search ended at index 76, at /home/hasegawa/ros2_rolling/src/ros2/rcl/rcl/src/rcl/lexer_lookahead.c:233
[DEBUG] [1611907728.783949213] [rcl]: Initializing node 'talker' in namespace ''
[DEBUG] [1611907728.784024386] [rcl]: Using domain ID of '0'
1611907728.801432 [0]     talker: using network interface enp0s25 (udp/192.168.11.2) selected arbitrarily from: enp0s25, docker0, ppp0
[DEBUG] [1611907728.865403015] [rcl]: Initializing publisher for topic name '/rosout'
[DEBUG] [1611907728.865498505] [rcl]: Expanded and remapped topic name '/rosout'
[DEBUG] [1611907728.910636385] [rcl]: Publisher initialized
[DEBUG] [1611907728.910692224] [rcl]: Node initialized
~~
^C
~~
[DEBUG] [1611907335.538527998] [rcl]: Finalizing node
^C
[INFO] [1611907336.001941525] [rclcpp]: signal_handler(signal_value=2)
[DEBUG] [1611907336.001980227] [rclcpp]: signal_handler(): SIGINT received, notifying deferred signal handler
[DEBUG] [1611907336.002017219] [rclcpp]: deferred_signal_handler(): woken up due to SIGINT or uninstall
[DEBUG] [1611907336.002053580] [rclcpp]: deferred_signal_handler(): SIGINT received, shutting down
[DEBUG] [1611907336.002081745] [rclcpp]: deferred_signal_handler(): waiting for SIGINT or uninstall
^C
[INFO] [1611907336.278440925] [rclcpp]: signal_handler(signal_value=2)
[DEBUG] [1611907336.278507928] [rclcpp]: signal_handler(): SIGINT received, notifying deferred signal handler
[DEBUG] [1611907336.278543761] [rclcpp]: deferred_signal_handler(): woken up due to SIGINT or uninstall
[DEBUG] [1611907336.278556990] [rclcpp]: deferred_signal_handler(): SIGINT received, shutting down
[DEBUG] [1611907336.278577575] [rclcpp]: deferred_signal_handler(): waiting for SIGINT or uninstall
```
2. This behavior isn't not only stuck during finalizing, but it also failed to send messages.
3. This issue didn't happen in docker container. I used [ros:rolling](https://hub.docker.com/layers/ros/library/ros/rolling/images/sha256-d37347f1213be3522a86567076283c69b078ef5ac99906257fed35a9ebb50680?context=explore) and installed cyclonedds.
4. installing rolling with apt causes same problem.
5. I tried to build CycloneDDS with [official procedure](https://github.com/eclipse-cyclonedds/cyclonedds) and run the Hello world sample without using ROS2, but no communication took place. However, this sample was terminated by Ctrl+C.

- Each communication was checked by executing subscriber and wireshark. There were no received messages and no packets. 
- The following warnings appeared in each case when communication was not possible.
```
$ ./HelloworldPublisher
1611906361.432950 [0] Helloworld: using network interface enp0s25 (udp/192.168.11.2) selected arbitrarily from: enp0s25, docker0, ppp0
 (^^^^ same as ros2 run demo_nodes_cpp talker)
=== [Publisher]  Waiting for a reader to be discovered ...
^C
(terminated)
```

ivanpauno, 2021-01-29:
> This issue didn't happen in docker container. I used ros:rolling and installed cyclonedds.

Ok, I use a docker container for ros 2 development.
Anyways this doesn't seem docker related (?), I guess it's just hard to trigger the deadlock situation.

ivanpauno, 2021-01-29:
I have tried to analyze the backtrace provided by @hsgwa, it seems that:

- `recv_thread()` (thread 9) is blocked in [do_packet](https://github.com/eclipse-cyclonedds/cyclonedds/blob/b84c035ce0292cd76358475f069da4aec1540bb2/src/core/ddsi/src/q_receive.c#L3500), which IIUC is retrying `ddsrt_recvmsg()` if an interrupt happens (see [here](https://github.com/eclipse-cyclonedds/cyclonedds/blob/aeb281d1e227e56799ef4671235679120ba66b3f/src/core/ddsi/src/ddsi_udp.c#L89-L91)).
  Maybe the interrupt is happening exactly in that `ddsrt_recvmsg()` call, it's retrying and that thread isn't receiving messages anymore for some reason (and thus, `ddsrt_recvmsg()` blocks forever).
  @hsgwa can you check if in your case " thread 9" is always reiterating in [that loop](https://github.com/eclipse-cyclonedds/cyclonedds/blob/aeb281d1e227e56799ef4671235679120ba66b3f/src/core/ddsi/src/ddsi_udp.c#L89-L91) without continuing?
  I see there's a `rtps_keepgoing` flag used in some other places, maybe that should be used there too (?).
- Thread 1 is destroying a node, which is destroying in this case a participant and then waiting to join the receiver threads (see [here](https://github.com/eclipse-cyclonedds/cyclonedds/blob/b84c035ce0292cd76358475f069da4aec1540bb2/src/core/ddsi/src/q_init.c#L1761)).

cc @eboasson , this may ring a bell to you.

fujitatomoya, 2021-01-30:
I cannot reproduce this issue so far with following environment.

- Intel(R) Core(TM) i7-3770 CPU @ 3.40GHz / Ubuntu20.04
- Raspberry Pi 4 Model B Rev 1.2 / Ubuntu20.04
- Docker containers based on ubuntu:20.04

could be related to network environment?

eboasson, 2021-01-30:
It is hanging because it is not succeeding in waking up a receive thread by sending a datagram to it: in principle, Cyclone can have any number of threads for receiving data from the network, and each thread handles either a single socket or multiple sockets. If it is the former, it enters a blocking wait, and then termination requires receiving a datagram to unblock the `recvmsg()` call. For those threads it sends itself a minimalistic datagram and retries every second.

From the stack traces, you can tell it is one of those threads, and you can also see its timed-event handling being in a state consistent with those retries:
```
#2  __pthread_cond_timedwait (cond=0x55a37be0a470, mutex=0x55a37be0a448, abstime=0x7fcf40a014d0) at pthread_cond_wait.c:656
#3  0x00007fcf432b60fb in ddsrt_cond_waituntil (cond=0x55a37be0a470, mutex=0x55a37be0a448, abstime=1611795496341632475) at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/sync/posix/sync.c:116
#4  0x00007fcf432b61bd in ddsrt_cond_waitfor (cond=0x55a37be0a470, mutex=0x55a37be0a448, reltime=999899872) at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/sync/posix/sync.c:137
#5  0x00007fcf432726a1 in xevent_thread (xevq=0x55a37be0a3f0) at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/core/ddsi/src/q_xevent.c:1286
```

On Linux, by default, it creates 3 threads: one that handles all discovery traffic (multicast+unicast, so 2 sockets and exempt from this termination procedure), one that handles unicast application data (single socket), and one that handles multicast application data (also single socket). Chances are it is the multicast one here, but I can't be quite certain.

Not being able to send datagrams to oneself is probably a network configuration issue, but I don't know what bit of is causing the problem. So it is probably sensible to avoid having to do this by not having any receive thread handling a single socket. For that, set [Internal/MultipleReceiveThreads](https://github.com/eclipse-cyclonedds/cyclonedds/blob/master/docs/manual/options.md#cycloneddsdomaininternalmultiplereceivethreads) to `false`.

iuhilnehc-ynos, 2021-01-31:
I can not reproduce this issue too, but I have a question about the implementation.

Could `Thread 1`
```
#4  0x00007fcf43250bfc in rtps_stop :1760
   rtps_term_prep -> trigger_recv_threads
```
calling `trigger_recv_threads`,
https://github.com/eclipse-cyclonedds/cyclonedds/blob/81666fd987cbb2a7a61f24c96be923c0cd1b2995/src/core/ddsi/src/q_receive.c#L3467-L3472

to make sure `Thread 9` 
```
#2  0x00007fcf432b52fc in ddsrt_recvmsg (sock=8, msg=0x7fcf2b7fd1b0, flags=0, rcvd=0x7fcf2b7fd188) at /home/hasegawa/ros2_rolling/src/eclipse-cyclonedds/cyclonedds/src/ddsrt/src/sockets/posix/socket.c:455
```
receive the triggered data? (It seems just UDP data without some confirmation.)

eboasson, 2021-01-31:
@iuhilnehc-ynos yes that's the path, at least for the initial "trigger" message. https://github.com/eclipse-cyclonedds/cyclonedds/blob/81666fd987cbb2a7a61f24c96be923c0cd1b2995/src/core/ddsi/src/q_init.c#L778 schedules a periodic event to try again (https://github.com/eclipse-cyclonedds/cyclonedds/blob/81666fd987cbb2a7a61f24c96be923c0cd1b2995/src/core/ddsi/src/q_init.c#L784). The purpose of the packet is to unblock the `recvmsg` call, the content of the packet doesn't matter, it just needs to return from `recvmsg`.

UDP/IP is unreliable and so it retries until the threads terminate, but as far as I know, the loopback interface is effectively reliable provided the socket receive buffer has space. If there is something present in the receive buffer, `recvmsg` should return it (and no matter what the contents are, it is sufficient to make it terminate); if there's nothing present, surely this packet should be delivered.

I've seen it hang in this way before, even on machines doing absolutely nothing else (so no chance of socket buffer being full and the IP stack dropping the packet for that reason), but because of some routing issue that made it impossible for a process to send a packet to its own sockets. I have no clue why such a routing issue would exist ...

iuhilnehc-ynos, 2021-02-01:
@eboasson 

Thank you for your explanation.

What if the following cases,
- initial "trigger" lost.  (`recvmsg` not to unblock)
- and time is adjusted back. (qxev_callback with `wait_for_receive_threads_helper` not scheduled a periodic event)

hsgwa, 2021-02-01:
Thanks!

I reviewed network settings and found out that firewall was the cause.
This issue can be reproduced as following steps.

1. Select interface to Ethernet

   1. Prepare XML

      ```xml
      <CycloneDDS>
          <Domain>
              <General>
                  <NetworkInterfaceAddress>YOUR_ETHERNET_INTERFACE</NetworkInterfaceAddress>
              </General>
              <Tracing>
                  <Category>config</Category>
                  
              </Tracing>
          </Domain>
      </CycloneDDS>
      ```

   2. Set CYCLONEDDS_URI

      ```bash
      $ export CYCLONEDDS_URI=/path/to/xml
      ```

2. Enable firewall

   ```bash
   $ sudo ufw enable
   ```

3. Run demo

   ```bash
   $ ros2 run demo_nodes_cpp talker
   ```

I tried to change cyclonedds configure, and these are results.

- If loopback is used, it will communicate and exit normally.
- When Internal/multipleReceiveThread=false, it does not communicate, but exits normally.

I apologize if this has already been documented about FIrewall settings.

eboasson, 2021-02-02:
Thanks @hsgwa! That all makes sense. However, this:

> When Internal/multipleReceiveThread=false, it does not communicate, but exits normally.

has me worried a little bit and I'd like to confirm that the firewall was enabled when it failed to communicate. If that's the case, there's no reason for me to worry (*) because that makes perfect sense (there's no discovery when the firewall blocks the discovery packets ...) and matches what I get when I try the same.

(*) I don't think it is ideal that it defaults to using only *one* network interface. It makes more sense to default to both the ethernet and the loopback. But that's another story.