ros2/rcl #890: Entities destruction return code handle lead to memory leaks

jparisu, 2021-02-11:
First of all, I am not sure this is a real issue in RCL or a design decision. It just looks to me as a bad behaviour of RCL.

This issue is regarding the return code handle in RCL entities destruction. `Publisher` will be used in this issue as an example, but the same process could be seen in every other entity.

In file `src/ros2/rcl/rcl/src/rcl/publisher.c` the function `rcl_publisher_fini` works as follows:
1. argument check
2. destroy rmw publisher
3. deallocate rcl publisher
The problem comes when the function `rmw_destroy_publisher` fails (`ret != RMW_RET_OK`). In this case, it means that the rmw entity has not been completely destroyed (at least, this would be the expected behavior in case the destruction fails). And this code error is used to set the destruction as `RCL_RET_ERROR` but it deallocates the `rcl_publisher` anyway, an thus, it left some memory leak with the entities below that will not be possibly recovered.

I would like to know if this is the expected behaviour for the RCL destruction (in case it is, I guess all ROS entities must be destroyed even when a lower entity fails, and I would like to know if there are design information regarding this behavior so I can apply it to RMW. But even `rcl_publisher` do not deallocate the entity in case the `node` is not valid).

P.D.: I got to this issue by running the RCL tests, in particular `test_mock_publisher_fini_fail`, where the mock do not allow to destroy correctly the lower entity. This could lead to a bigger issue than a memory leak when it interferes with the destruction of a whole context at the end of the test.
There are ways for this test to run correctly without memory leak and using this mock, but not if `rcl_publisher` is erased before correctly remove the lower entities.

If this happens to be a real issue in RCL, and not the expected behaviour, I would be happy to open a Pull Request and display my alternatives. 
So do not hesitate to contact me with every comment

## Bug report

**Required Info:**

- Operating System:
Ubuntu 20.04
- Installation type:
source
- Version or commit hash:
rolling
- DDS implementation:
Test done with Fast-RTPS. But it will happen in every implementation

#### Steps to reproduce issue

```
colcon test --event-handlers console_direct+ --packages-select rcl --ctest-args -R test_publisher__rmw_fastrtps_cpp
```

#### Expected behavior
Every Entity is erased correctly at the end of the test.

#### Actual behavior
In `test_mock_publisher_fini_fail` the `rmw_publisher` is not removed in the test or in `TearDown`.

## Feature request
Stop the destruction execution in case a lower entity fails in destruction.

#### Feature description
In case an RMW entity fails in destroy process, the RCL entity must stop the destruction execution so no memory leaks are left.

#### Implementation considerations
Different Entity destructions may handle the return code errors of lower entities destruction, and should stop in fail case.

fujitatomoya, 2021-02-12:
i think that's been considered before...i am not saying the current interfaces are perfect, probably there would be some additional error handlings, but i think eventually there will be something like current implementation to handle unexpected errors.

> I would be happy to open a Pull Request and display my alternatives.

👍 that would be appreciated!

> Different Entity destructions may handle the return code errors of lower entities destruction, and should stop in fail case.

i would like to know what is gonna happen after `stop`.

jparisu, 2021-02-18:
Hi @fujitatomoya 

Sorry for the delay.

It is true that there is not a perfect way to handle this behavior,
1. In case the destruction continues, the lower entity will remain and a memory leak will occur. It should not be such important if it wasn't because of ROS 2 common use case is the repeatedly creation and deletion of entities.
2. In case the destruction stops there are two problems:
2.1. The graph has been updated already when this error will occur. This is a minor problem, as if the destruction works successfully this is the expected behavior and it will not harm the rest of the network.
2.2 The higher entity is not deleted. This is a problem, but at least it is not a memory leak, so this memory space could be deleted afterwards.

Following the DDS specification https://www.omg.org/spec/DDS/1.4/PDF the destruction of entities must rely in the correct destruction of lower entities (you cannot delete a `Publisher` with `DataWriters` or a `DomainParticipant` with `Publishers`), and thus I see this design a better approach to this problem. (Fast-DDS follows such policy).

In order to show a possible implementation, I have uploaded a PR https://github.com/ros2/rcl/pull/891 so you can check the main idea and probe its results (rcl tests works correctly and a memory leak in test `test_mock_publisher_fini_fail` has been fixed).

This implementation only changes the `publisher` behavior because this was a first approach. In case you finally decide to go with it I could change the other entities as well and look for better error handlers.

jparisu, 2021-02-18:
Such change will imply to change many other higher and lower interfaces as `rclcpp` and `rmw`.
And, as it is a major change with few repercussions I do not coherent to keep this issue opened.

Thank you, and I will be glad to know any news in the topic.

closed 2021-02-18 by jparisu