ros2/rcl #898: Add `sleep_for` and `sleep_until` for `rcl_clock_t` as per "Clock and Time" design

emersonknapp, 2021-03-09:
## Feature request

#### Feature description

The design article [Clock and Time](http://design.ros2.org/articles/clock_and_time.html#ros-time) specifies that the ROS clock API is to have `sleep_for` and `sleep_until` methods. These are not yet present (that I know of).

For context - it seems that this feature would be the best option to use for controlling rosbag2 playback time, instead of using `std::this_thread::sleep_until` in the message playback loop, so that it is more possible to handle time source control such as pausing and alteration of playback speed - as well as listening to an external time source like Gazebo.

#### Implementation considerations

Likely the existing timer code gives us most of the tools we need to implement this, it uses time sources correctly.

fujitatomoya, 2021-03-10:
this sounds reasonable to me, though it does seem to be really common use case. i wonder there's been similar discussion before... 🤔

jacobperron, 2021-03-12:
Related ticket: https://github.com/ros2/rclcpp/issues/465

Also a long discussion in https://github.com/ros2/rclcpp/issues/1248. TL;DR, I think the consensus is that we should have a "sleep_until" function that operates on a ROS time object, though the discussion was focused on rclcpp. I don't know how much development overhead is involved in abstracting this feature to `rcl`; I guess it may be a bit trickier without the convenience of some C++ features.

emersonknapp, 2021-03-12:
I really only need it in C++ right now for this project, but I'm generally inclined to want `rclpy` (et. al) to have features, too.

paulsammut, 2021-07-29:
I just want to be able to sleep for x number of time :(

sloretz, 2021-11-30:
Sleeping is thread specific, and threads are language specific, so I think it makes sense to implement sleep methods in the language specific client libraries instead of `rcl`.

This has been implemented in rclcpp - see the related ticket #1730

clalancette, 2022-06-21:
This was implemented in both rclcpp and rclpy, and it seems like we are not going to do anything further in rcl itself.  So closing this out.

closed 2022-06-21 by clalancette