ros2/rcl #906: Allow zero initialize drcl_arguments_t to be fini'd without error

sloretz, 2021-03-29:
`rcl_arguments_t` errors if it's fini'd twice, which means it errors if the `rcl_arguments_t` object is zero initialized.

https://github.com/ros2/rcl/blob/b7784eb5d2fbff33053b69f6b9f9d3b703f9f2a7/rcl/src/rcl/arguments.c#L995-L996

Other rcl types, like `rcl_node_t`, do nothing when the object is zero initialized.

https://github.com/ros2/rcl/blob/b7784eb5d2fbff33053b69f6b9f9d3b703f9f2a7/rcl/src/rcl/node.c#L368-L371

It's easier to use a type that allows fini'ing twice, because cleanup logic can be setup unconditionally at the beginning of the type creation instead of conditionally only when the type creation succeeds.

hidmic, 2021-03-29:
I vaguely recall some discussion about finalization during the `rmw` API review. It is true that allowing double fini'ing simplifies cleanup, but outside that use case, it indicates bogus code IMHO. That's why I've been always inclined towards not allowing it. It's easier to spot bad lifecycle management that way.

fujitatomoya, 2021-03-29:
> It's easier to use a type that allows fini'ing twice, because cleanup logic can be setup unconditionally at the beginning of the type creation instead of conditionally only when the type creation succeeds.

i am inclined to this a little bit. either way, i believe that it would be nice to keep the behavior consistent.