ros2/rcl #920: Imported target "rcl::rcl" includes non-existent path

norro, 2021-06-08:
## Bug report

The galactic binary package for Windows seems to be polluted with a path from the original build environment, `"C:/ci/ws/install/include"`, resulting in downstream package builds to fail, see details below.

**Required Info:**

- Operating System: Windows 10
- Installation type: binaries
- Version: 9.1.0
- Client library: rcl

#### Steps to reproduce

Build a downstream package from source on Windows against the rcl binary package, e.g., build [rclc](https://github.com/ros2/rclc/). 

#### Actual behavior

CMake of downstream package fails with the following error:

```
CMake Error in CMakeLists.txt:
  Imported target "rcl::rcl" includes non-existent path

    "C:/ci/ws/install/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.
```

#### Additional information

When checking `rclExport.cmake` I can see that the binary package indeed does export the path `"C:/ci/ws/install/include"`:

[image]

norro, 2021-06-08:
Same with `rcl_lifecycle` and `rclcpp`, yet not with `rclcpp_lifecycle`.

allenh1, 2021-06-08:
I've seen this on foxy (on Linux), actually (we build a release on ubuntu 18.04 from source, so I assumed I just had the wrong colcon settings or something). Would be great to know what causes this.

rlkandela, 2021-06-11:
Seems to be the same as [this](https://github.com/ros2/rclcpp/issues/1688). There I proposed a workaround that solves it.

ivanpauno, 2021-07-26:
> I've seen this on foxy (on Linux), actually (we build a release on ubuntu 18.04 from source, so I assumed I just had the wrong colcon settings or something). Would be great to know what causes this.

See https://github.com/ros2/rclcpp/issues/1688#issuecomment-886740581.

clalancette, 2022-06-21:
While we haven't fixed all of these issues in all packages (there are still one or two more fixes to go), we have fixed this in Humble and Rolling for `rcl::rcl`.  So closing this out.

closed 2022-06-21 by clalancette