ros2/rcl #922: fini method on subscriber can't forward non binary returns from RMW

bjv-capra, 2021-06-30:
- Installation type:
  - from source
- Version or commit hash:
  - Foxy
- DDS implementation:
  - microxrcedds

Hi, on https://github.com/micro-ROS/micro_ros_setup/issues/347 I try to call the subscription `_fini` method when the micro-ROS agent disconnects. When the agent disconnects, my rmw will return `RMW_RET_TIMEOUT` which is actually to indicate that the memory has been deallocated on the client, but on the agent's memory (this is fine). However, the RCL will return `RCL_RET_ERROR` upon any non 0 return.

Is this an intended behavior? Or could we patch this so that rcl can forward the `RMW_RET_TIMEOUT`. The case on the matter is at https://github.com/ros2/rcl/blob/master/rcl/src/rcl/subscription.c#L175` however, it probably also applies to the topics and services.

jacobperron, 2021-07-22:
In short, this is the intended behavior as documented:

https://github.com/ros2/rcl/blob/80efa3fa0861da45c1f81622763d06fc1f76f129/rcl/include/rcl/subscription.h#L185-L189

I'd argue, the fact that `rmw_destroy_subcription` is returning `RMW_RET_TIMEOUT` is violating the rmw API:

https://github.com/ros2/rmw/blob/dcd8f836e6c2545b39c1540963e175f06625e2f0/rmw/include/rmw/rmw.h#L1041-L1045

If we want to add a new return code to `rcl_subscription_fini`, I think we should first update the rmw API (which is a separate discussion).

---

Naively, in your scenario, if everything is fine then maybe the rmw should return `RMW_RET_OK`?

clalancette, 2022-06-21:
No response in almost a year, so I'm going to close this out.  If there is still something to follow-up or discuss here, please feel free to reopen.

closed 2022-06-21 by clalancette