ros2/rcl #925: intra-process communication with serialized message

Jackey-Huo, 2021-07-12:
Hi all,

I'm work on a embedded linux system with 4-core A53 platform, and I'm planning to use ros2 to implement the intra-process communication.

For some reason, I'm not willing to use the default ros msg generation mechanism, but some other IDL language generator (like protobuf, flatsbuffer and so on), so I'd like to pass a serialized message directly to to ros publish function. But then, in the source code, I found the serialized message publish for intra-process is not implemented yet
https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/publisher.hpp#L466

So is there any schedule to implement the serialized publish recently? or is there any alternative method to achieve my goal?

clalancette, 2021-07-12:
Please don't open the same issue on multiple repositories; it is redundant and can split the conversation.  I'm closing this one in favor of https://github.com/ros2/rclcpp/issues/1710

closed 2021-07-12 by clalancette

Jackey-Huo, 2021-07-20:
Sorry for that, thank you so much for the bother