ros2/rcl #926: Expiring multiple action results causes crash

Scotty1701, 2021-07-13:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - Chocolatey Install
- Version or commit hash:
  - Foxy
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp_action

#### Steps to reproduce issue

Start the example fibonacci action server:
`ros2 run action_tutorials_cpp fibonacci_action_server`  

Send 2 goals to the server in quick succession:
`ros2 action send_goal /fibonacci action_tutorials_interfaces/action/Fibonacci "{}"`  

Wait 15 minutes (or however long the result_timeout for the action server is set for).

#### Expected behavior
Action Server continues running
#### Actual behavior
Action Server crashes with exception code `0xc0000409` 
#### Additional information
Sending one goal, waiting for the result to expire, and then sending another does not cause a crash, only when there are multiple cached goals does a crash occur. To avoid a 15 minute wait before a crash, you can set the `result_timeout.nanoseconds` field of an `rcl_action_server_options_t` object and pass that to the `rclcpp_action::create_server` function.

fujitatomoya, 2021-09-07:
I see this is the problem in theory and logically, but cannot make it happen.

closed 2021-09-14 by clalancette

KavenYau, 2021-09-17:
```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007f8bb0ac3859 in __GI_abort () at abort.c:79
#2  0x00007f8bb0b2e3ee in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7f8bb0c5807c "*** %s ***: terminated\n") at ../sysdeps/posix/libc_fatal.c:155
#3  0x00007f8bb0bd0b4a in __GI___fortify_fail (msg=msg@entry=0x7f8bb0c58064 "stack smashing detected") at fortify_fail.c:26
#4  0x00007f8bb0bd0b16 in __stack_chk_fail () at stack_chk_fail.c:24
#5  0x00007f8bb0776571 in rclcpp_action::ServerBase::execute_check_expired_goals() () from /home/kaven/ros2_ws/install/rclcpp_action/lib/librclcpp_action.so
#6  0x7deabfa5e2ea4802 in ?? ()
#7  0x3707acb461434114 in ?? ()
#8  0x162168ace7f673b2 in ?? ()
#9  0x7d9d995e8448ee4d in ?? ()
#10 0x3a0160f661434115 in ?? ()
#11 0xa5dfa2a0f83d899f in ?? ()
#12 0x58790a2639635c28 in ?? ()
#13 0x09444a6861434117 in ?? ()
#14 0x61dc685b9a6a2f3b in ?? ()
#15 0x06d52596f0b1e101 in ?? ()
#16 0x0c416ec761434118 in ?? ()
#17 0x957f8a9bd93a0845 in ?? ()
#18 0xee55f19df09ba4a3 in ?? ()
#19 0x0f214afd61434119 in ?? ()
#20 0x28243330a9c0ca32 in ?? ()
#21 0x760d2152abb21d59 in ?? ()
#22 0x12239e8d6143411a in ?? ()
#23 0x6fd24b3bc403950a in ?? ()
#24 0x64edc8aa0c34c025 in ?? ()
#25 0x151307016143411b in ?? ()
#26 0x00007fff333a5bd0 in ?? ()
#27 0x00007fff333a5b80 in ?? ()
#28 0x00007f8bb17176d6 in rclcpp::executors::SingleThreadedExecutor::spin() () from /home/kaven/ros2_ws/install/rclcpp/lib/librclcpp.so
Backtrace stopped: Cannot access memory at address 0x92933d130abe9902
```
I think I got this issue on ubuntu 20.04 too.

clalancette, 2021-09-17:
@KavenYau Can you confirm that the latest code with #931 applied fixes your problem?

KavenYau, 2021-11-18:
@clalancette Yes, with applying #931, it was fixed.

PR #931 (2021-09-14): fix expired goals capacity of action server
fix #926

This fix is important for action server nodes especially on Windows.