ros2/rcl #950: can_loan_messages can never be true during test

fujitatomoya, 2021-11-04:
## Bug report

- Operating System:
  - Ubuntu20.04
- Installation type:
  - source build
- Version or commit hash:
  - https://github.com/ros2/ros2/commit/4a36f319afcc208deae3c373493ef6d9feda1324
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rcl

#### Steps to reproduce issue

```
colcon test --packages-select rcl
```

#### Expected behavior

#### Actual behavior

https://github.com/ros2/rcl/blob/d8817fdd24c33b5e02786663f804a6240746c751/rcl/test/rcl/test_publisher.cpp#L358-L370

the above if statement should be true via rmw_fastrtps, and test code should be executed.

#### Additional information

I think Fast-RTPS is the only one implementation that can enable `can_loan_messages` in this test case, since RTI returns `false` and Cyclonedds needs iceoryx.

fujitatomoya, 2021-11-04:
@Barry-Xu-2018 @iuhilnehc-ynos could you take a look at this?

Barry-Xu-2018, 2021-11-04:
Related code on whether support loaned message for rmw_fastrtps_cpp

https://github.com/ros2/rmw_fastrtps/blob/32ce21d2d0250a46b9eeda9df1cdcd76503a5823/rmw_fastrtps_cpp/src/publisher.cpp#L298-L299

Existing problems:
1. Message `String` is non-pod
https://github.com/ros2/rcl/blob/d8817fdd24c33b5e02786663f804a6240746c751/rcl/test/rcl/test_publisher.cpp#L343-L344  

2. data_sharing ins't enabled by default for fast-dds. Should use config file to enable data_share for test topic.

fujitatomoya, 2021-11-04:
> Message String is non-pod

i was aware of this, so i tried with POD data type.

> data_sharing ins't enabled by default for fast-dds. Should use config file to enable data_share for test topic.

okay, that was my misunderstanding. i thought that is enabled by default.

fujitatomoya, 2021-11-04:
@Barry-Xu-2018 thanks for checking on this. i think this is not a bug, so i just gonna close this one.

closed 2021-11-04 by fujitatomoya