ros2/rcl #954: Assign parameter to any node doesn't work, if the node has namespace

amfern, 2021-12-07:
## Bug report
Assign parameter to any node doesn't work, if the node has namespace.

**Required Info:**

- Operating System:
  - ubuntu:20.04
- Installation type:
  -  binaries
- Version or commit hash:
  - ros-galactic
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
create config.yaml file with the following content
```
/*:
  parameter_blackboard:
    ros__parameters:
      miau: "miau_config"
```
/opt/ros/galactic/lib/demo_nodes_cpp/parameter_blackboard --ros-args -r __ns:=/aaaa --params-file config.yaml
ros2 param list

#### Expected behavior
ros2 param list should list the parameter miau, expected output is
```
/aaaa/parameter_blackboard:
  miau
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  use_sim_time
```

#### Actual behavior
miau parameter isn't added to the node, ros2 param list returns
```
/aaaa/parameter_blackboard:
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  use_sim_time
```

Barry-Xu-2018, 2021-12-08:
https://github.com/ros2/rclcpp/pull/1280  

It seems above PR implements this functionality. But it isn't finished.   
Besides, it is better to move this issue to rclcpp.

amfern, 2021-12-08:
Why it rclcpp issue, isn't the parameter and yaml parsing happening in rcl?

Barry-Xu-2018, 2021-12-08:
> Why it rclcpp issue, isn't the parameter and yaml parsing happening in rcl?

YAML parsing is in rcl. But further parsing (e.g. /**) is done in rclcpp.    

@iuhilnehc-ynos submit PR to fix this bug. I think you can try it.   
https://github.com/ros2/rclcpp/pull/1839

amfern, 2021-12-11:
[this](https://github.com/ros2/rclcpp/pull/1839) works for me, it fixes this issue

closed 2022-07-06 by fujitatomoya