ros2/rcl #962: Remap fails for topic which contains `/0`

oceanusxiv, 2022-02-19:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - Galactic
- DDS implementation:
  - Default
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
Simply call
```
ros2 run demo_nodes_cpp talker --ros-args -r /hello/0/hello:=/goodbye
```

#### Expected behavior
The remap should just work.
#### Actual behavior
```
terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
  what():  failed to initialize rcl: Couldn't parse remap rule: '-r /hello/0/hello:=/goodbye'. Error: Expecting token or wildcard, at /tmp/binarydeb/ros-galactic-rcl-3.1.2/src/rcl/arguments.c:1170, at /tmp/binarydeb/ros-galactic-rcl-3.1.2/src/rcl/arguments.c:371
```
#### Additional information
There seems to be something odd with the rcl lexer. There's no rule that topic names can't have `/0` in them and yet this causes an error when remapping them.

clalancette, 2022-02-19:
> There seems to be something odd with the rcl lexer. There's no rule that topic names can't have `/0` in them and yet this causes an error when remapping them.

Actually I don't think that is true.  Parts of topic names can't start with a numeric character, as per http://design.ros2.org/articles/topic_and_service_names.html#name-tokens .  The fact that the system allows you to create `/hello/0/hello` at all is the bug here, I think.

oceanusxiv, 2022-02-20:
> Actually I don't think that is true. Parts of topic names can't start with a numeric character, as per http://design.ros2.org/articles/topic_and_service_names.html#name-tokens . The fact that the [...]

Oh interesting, so it's actually not allowed in ros2. I was working with ignition gazebo and one of their default output topics outputs a `/0/`, per 
https://github.com/ignitionrobotics/ign-gazebo/blob/7901903d06a24b9a62d2e584241cfd4d95798f39/src/systems/joint_position_controller/JointPositionController.hh#L39
, which isn't ideal when bridged to ros2.

Guess it's just one more compatibility issue between ignition and ros2. Thanks for clarifying, let's see if I can raise it up with them instead.

closed 2022-02-20 by oceanusxiv