ros2/rcl #971: Inaccurate null validation for timer

NoyZuberi, 2022-03-20:
Hi,

In the file timer, there are several null checks for the argument timer 
`RCL_CHECK_ARGUMENT_FOR_NULL(timer, RCL_RET_INVALID_ARGUMENT);` 
But, as I understand this check not really a good validation since in here it's only verify whether the pointer for timer argument is null but not it's field `struct rcl_timer_impl_t * impl;` 

In my implementation, using this library, I had issues sending null timer to `rcl_timer_exchange_period` and didn't understand exactly why, so I added locally another check for all those functions that are using the timer argument 
`RCL_CHECK_FOR_NULL_WITH_MSG(timer->impl, "timer is invalid", return RCL_RET_INVALID_ARGUMENT);` and that caught those time I sent null timer.

I think it is a fix that can help others that might have the same problems as I do.
Thanks.

clalancette, 2022-03-21:
It sounds like a reasonable check to make.  If you'd like to open a pull request, we'd be happy to review it.  Thanks!

NoyZuberi, 2022-03-21:
Sure, will open.
Thanks.

NoyZuberi, 2022-03-23:
Something that I didn't understand, maybe I am missing something.
I tried to create pull request for ros2/rcl, so I created fork for it, but for some reason it created me a fork for micro-ros/rcl.
I know that ros2/rcl is a fork for micro-ros/rcl, but why can't I edit the fork ros2/rcl?

Regardless of that I still created PR for micro-ros/rcl.

Thanks.

jacobperron, 2022-05-03:
I think this was fixed in #973. Please correct me if I'm wrong.

closed 2022-05-03 by jacobperron