ros2/rcl #984: Add some sort of identifying information to the log message when duplicate nodes are created

jlack1987, 2022-05-15:
When creating duplicate nodes(in my experience from rclcpp) a message is printed to the console,
```
[rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
```
It can be very difficult from this message to figure out where the offending line(s) of code may be in our very large stack of code. Is there any way to get some sort of identifying information into this? I took a brief glance at the code and even putting the `logger_name` in the message would be really helpful. If the node name is available that would be optimal, but anything to help the developer figure out where it's coming from would be a step in the right direction. I'm willing to put in a fix for this if a dev can recommend something that would be acceptable.

It looks to me like the chunk of code where this originates from is,

```
if (rcutils_hash_map_key_exists(&__logger_map, &logger_name)) {
    // @TODO(nburek) Update behavior to either enforce unique names or work with non-unique
    // names based on the outcome here: https://github.com/ros2/design/issues/187
    RCUTILS_LOG_WARN_NAMED(
      "rcl.logging_rosout",
      "Publisher already registered for provided node name. If this is due to multiple nodes "
      "with the same name then all logs for that logger name will go out over the existing "
      "publisher. As soon as any node with that name is destructed it will unregister the "
      "publisher, preventing any further logs for that name from being published on the rosout "
      "topic.");
    return RCL_RET_OK;
}
```

fujitatomoya, 2022-05-16:
> putting the logger_name in the message would be really helpful.

:+1: that sounds reasonable to me.

CursedRock17, 2023-09-23:
This issue can be closed if the PR was enough.

fujitatomoya, 2023-09-23:
@CursedRock17 good eye 👁️ closing.

closed 2023-09-23 by fujitatomoya