ros2/rcl #986: Do not require the node argument in publisher and subscription fini functions

nnmm, 2022-05-19:
## Feature request
Do not require the node argument in publisher and subscription fini functions.

#### Feature description
Passing in the node to `rcl_subscription_fini` and `rcl_publisher_fini` allows passing in a node that's not the same node as the one used in the init function.

The publisher and subscription could instead store the node pointer given to the init function and use it later in the fini function. This would also make the `rclrs` client library slightly more elegant.

#### Implementation considerations
I assume changing the signatures of the fini functions would cause too much breakage (although I'm not really sure about that, there probably aren't many users of the `rcl` API in the wild). So we could add a `rcl_subscription_fini_2`/`rcl_publisher_fini_2` function.

nnmm, 2022-05-19:
cc @sloretz, you mentioned this yesterday. I can work on it, but can you remind me of what you think regarding changing signatures?

wjwwood, 2022-05-31:
The reason for having the node argument passed via this fini function is to make it clear (by contract) that the node needs to be finalized after the subscription/publisher.

I would not remove this argument, but I do support storing and checking that the correct node was used in fini.

fujitatomoya, 2023-01-05:
This issue has no activity more than 6 months, i will go ahead to close this.

closed 2023-01-05 by fujitatomoya