ros2/rcl #989: Guard condition not getting marked as ready in waitset (possible race condition?)

hoffmann-stefan, 2022-06-29:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04 in WSL2 on Windows 10 20H2 (Build 19042.1706)
  - uname -a: `Linux Client-291 5.10.60.1-microsoft-standard-WSL2 #1 SMP Wed Aug 25 23:20:18 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`
- Installation type:
  - binaries
- Version or commit hash:
```bash
$ apt-cache policy ros-foxy-rcl
ros-foxy-rcl:
  Installed: 1.1.13-1focal.20220209.142931
  Candidate: 1.1.13-1focal.20220209.142931
  Version table:
 *** 1.1.13-1focal.20220209.142931 500
        500 http://packages.ros.org/ros2/ubuntu focal/main amd64 Packages
        100 /var/lib/dpkg/status
  ```
- DDS implementation:
  - rmw_fastrtps_cpp
- Client library (if applicable):
  - rcldotnet (my "feature" branch: https://github.com/ros2-dotnet/ros2_dotnet/pull/94)

#### Steps to reproduce issue

Using this branch that adds an example to reproduce this run the follwoing command:
https://github.com/hoffmann-stefan/ros2_dotnet/commits/reproduce_gaurd_condition_not_triggerd
```bash
$ ros2 run rcldotnet_examples rcldotnet_guard_condition
```

Here is the console application that causes this: (logging is was added to `rcldotnet` in the linked branch)

```C#
using System;
using System.Threading;
using System.Threading.Tasks;

using ROS2;

namespace ConsoleApplication
{
    public static class RCLDotnetGuardCondition
    {
        static GuardCondition _guardCondition = null;

        public static void Main(string[] args)
        {
            RCLdotnet.Init();

            var node = RCLdotnet.CreateNode("guard_condition_example");

            _guardCondition = node.CreateGuardCondition(GuardConditionTriggered);

            Console.WriteLine(RCLdotnet.GetRMWIdentifier());
            Console.WriteLine($"ThreadId: {Environment.CurrentManagedThreadId}");

            _guardCondition.Trigger();

            RCLdotnet.Spin(node);
        }

        static void GuardConditionTriggered()
        {
            _ = TriggerAfterSomeTime();
        }

        static async Task TriggerAfterSomeTime()
[... 89 lines cut ...]
2022-06-29T11:18:14.0735032+02:00 ThreadId: 1 RCLdotnet.SpinOnce(): add guard condition to waitset
2022-06-29T11:18:14.5739703+02:00 ThreadId: 1 RCLdotnet.SpinOnce(): timeout wait
2022-06-29T11:18:14.5741613+02:00 ThreadId: 1 RCLdotnet.SpinOnce(): add guard condition to waitset
2022-06-29T11:18:15.0745775+02:00 ThreadId: 1 RCLdotnet.SpinOnce(): timeout wait
2022-06-29T11:18:15.0747819+02:00 ThreadId: 1 RCLdotnet.SpinOnce(): add guard condition to waitset
2022-06-29T11:18:15.5753164+02:00 ThreadId: 1 RCLdotnet.SpinOnce(): timeout wait
2022-06-29T11:18:15.5755453+02:00 ThreadId: 1 RCLdotnet.SpinOnce(): add guard condition to waitset
2022-06-29T11:18:16.0760885+02:00 ThreadId: 1 RCLdotnet.SpinOnce(): timeout wait
2022-06-29T11:18:16.0763506+02:00 ThreadId: 1 RCLdotnet.SpinOnce(): add guard condition to waitset
2022-06-29T11:18:16.5771038+02:00 ThreadId: 1 RCLdotnet.SpinOnce(): timeout wait
2022-06-29T11:18:16.5773719+02:00 ThreadId: 1 RCLdotnet.SpinOnce(): add guard condition to waitset
2022-06-29T11:18:17.0777013+02:00 ThreadId: 1 RCLdotnet.SpinOnce(): timeout wait
2022-06-29T11:18:17.0779539+02:00 ThreadId: 1 RCLdotnet.SpinOnce(): add guard condition to waitset
2022-06-29T11:18:17.5784777+02:00 ThreadId: 1 RCLdotnet.SpinOnce(): timeout wait
2022-06-29T11:18:17.5787394+02:00 ThreadId: 1 RCLdotnet.SpinOnce(): add guard condition to waitset
2022-06-29T11:18:18.0790355+02:00 ThreadId: 1 RCLdotnet.SpinOnce(): timeout wait
2022-06-29T11:18:18.0791211+02:00 ThreadId: 1 RCLdotnet.SpinOnce(): add guard condition to waitset

```

#### Additional information

I'm not sure if this is a supported pattern to use guard_conditions in combination with waitsets.
Before further investigation I would like to ask if I am doing something wrong while adding guard conditions to rcldotnet? (see this commit: https://github.com/hoffmann-stefan/ros2_dotnet/commit/b7713c5084367a998e6e04a109b0990414c56110)
I didn't find documentation for this, so if I'm doing someting wrong maybe we can extend the header documentation?

ivanpauno, 2022-08-18:
This repo is for `rcl` issues, not for `rcldotnet`.
If you believe an issue in `rcldotnet` is caused by an issue in `rcl`, please provide a minimal reproducible example of the bug written in C (or C++) using `rcl` directly.

Feel free to update the description and reopen or open a new issue when you have that.
Thanks!

ivanpauno, 2022-08-18:
> I'm not sure if this is a supported pattern to use guard_conditions in combination with waitsets.

I have never used `rcldotnet`, but the example you provided seems like something that it should work if the client library is implemented correctly.

ivanpauno, 2022-08-18:
> I have never used rcldotnet, but the example you provided seems like something that it should work if the client library is implemented correctly.

I don't know how async in dotnet works as well, so there may also be a problem with that ...

closed 2022-08-18 by ivanpauno

hoffmann-stefan, 2022-08-22:
Got this working with switching the ROS2 middleware to `cyclonedds`, so at least my branch for rcldotnet can't be completely wrong.

Was hoping maybe something obvious for people knowing rcl would be found while glimpsing over the linked source code, but It seems that there is no way around a c/c++ minimal example. Though I can't promise when/if I get to this.

As for C# async this should be like c++ promise/features with additional syntactic sugar to avoid callback hell. The code after the `Task.Delay` continues in a background thread that is managed by the dotnet runtime. Though I don't know exactly which implementation details this has that may interfere with the multi-threading synchronization used by rcl/rmw.