ros2/rcl #1001: uninformative crash and error message for invalid parameter override

christianrauch, 2022-08-18:
## Bug report

Setting an invalid parameter override crashes the process and gives an uninformative error message.

**Required Info:**

- Operating System: Ubuntu 22.04
- Installation type: binaries
- Version or commit hash: 5.3.1
- DDS implementation: default
- Client library (if applicable): rcl

#### Steps to reproduce issue
1. `sudo apt install ros-humble-demo-nodes-cpp`
2. `ros2 run demo_nodes_cpp talker --ros-args -p bla`

#### Expected behavior

An error message or warning informing about the missing value assignment should be shown. I.e. the message should say that `-p bla` is not a valid parameter override, but `-p bla:=5` is.

Even better, the parameter `bla` should be set to uninitialised (`ParameterType::PARAMETER_NOT_SET`).

#### Actual behavior

Crash with:
```
[ERROR] [1660855233.229710370] [rcl]: Failed to parse global arguments
terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
  what():  failed to initialize rcl: Couldn't parse parameter override rule: '-p bla'. Error: Expected lexeme type (22) not found, search ended at index 3, at ./src/rcl/lexer_lookahead.c:239, at ./src/rcl/arguments.c:343
[ros2run]: Aborted
```

fujitatomoya, 2022-08-19:
I think that this is because application does not catch the exception from `rclcpp::init` (`rcl_init`)

```bash
# ros2 run demo_nodes_cpp add_two_ints_client 1 2 --ros-args -p bla
[ERROR] [1660931840.357026857] [rcl]: Failed to parse global arguments
terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
  what():  failed to initialize rcl: Couldn't parse parameter override rule: '-p bla'. Error: Expected lexeme type (22) not found, search ended at index 3, at /root/ros2_ws/colcon_ws/src/ros2/rcl/rcl/src/rcl/lexer_lookahead.c:239, at /root/ros2_ws/colcon_ws/src/ros2/rcl/rcl/src/rcl/arguments.c:343
[ros2run]: Aborted
```

this can be with following patch,

```patch
diff --git a/demo_nodes_cpp/src/services/add_two_ints_client.cpp b/demo_nodes_cpp/src/services/add_two_ints_client.cpp
index 8442801..b0cdf0d 100644
--- a/demo_nodes_cpp/src/services/add_two_ints_client.cpp
+++ b/demo_nodes_cpp/src/services/add_two_ints_client.cpp
@@ -43,7 +43,12 @@ int main(int argc, char ** argv)
   // Force flush of the stdout buffer.
   setvbuf(stdout, NULL, _IONBF, BUFSIZ);
 
-  rclcpp::init(argc, argv);
+  try {
+    rclcpp::init(argc, argv);
+  } catch (const std::exception & exception) {
+    RCLCPP_ERROR(rclcpp::get_logger("rclcpp"), exception.what());
+    return EXIT_FAILURE;
+  }
 
   auto node = rclcpp::Node::make_shared("add_two_ints_client");
 ```

```bash
# ros2 run demo_nodes_cpp add_two_ints_client 1 2 --ros-args -p bla
[ERROR] [1660931652.776928271] [rcl]: Failed to parse global arguments
[ERROR] [1660931652.776993285] [rclcpp]: failed to initialize rcl: Couldn't parse parameter override rule: '-p bla'. Error: Expected lexeme type (22) not found, search ended at index 3, at /root/ros2_ws/colcon_ws/src/ros2/rcl/rcl/src/rcl/lexer_lookahead.c:239, at /root/ros2_ws/colcon_ws/src/ros2/rcl/rcl/src/rcl/arguments.c:343
[ros2run]: Process exited with failure 1

```

clalancette, 2022-08-19:
> [rclcpp]: failed to initialize rcl: Couldn't parse parameter override rule: '-p bla'. Error: Expected lexeme type (22) not found, search ended at index 3, at /root/ros2_ws/colcon_ws/src/ros2/rcl/rcl/src/rcl/lexer_lookahead.c:239, at /root/ros2_ws/colcon_ws/src/ros2/rcl/rcl/src/rcl/arguments.c:343

I still think this error message isn't as useful as it could be.  It's exposing a low-level detail (the lexer parse error), rather than the higher-level "you provided an parameter argument that is missing :=", which is arguably more useful to the end-user.

iuhilnehc-ynos, 2022-09-09:
I'll use `RCUTILS_LOG_DEBUG_NAMED`(because some developers might need this kind of information.) instead of `RCL_SET_ERROR_MSG_WITH_FORMAT_STRING` in https://github.com/ros2/rcl/blob/392f0d3ae6a29d4b88689b954b14f19716e44233/rcl/src/rcl/lexer_lookahead.c#L239-L246, and then to call `RCL_SET_ERROR_MSG_WITH_FORMAT_STRING` with a user-friendly error message in the location calling `rcl_lexer_lookahead2_expect` if a failure happened. some locations like https://github.com/ros2/rcl/blob/ab4f1d6e7b6da6ecdf78d3d32c12a9c828c98bf1/rcl/src/rcl/arguments.c#L1014-L1017

abhishek47kashyap, 2025-11-28:
The error message originates from
https://github.com/ros2/rcl/blob/b7d6d69e670aa97bf69a6b92d12321ed31e68a4c/rcl/src/rcl/arguments.c#L352-L354
- `argv[i]` resolves to `-p`
- `argv[i + 1]` resolves to `bla` (per the steps-to-reproduce example)
- `prev_error_string.str` produces `Expected lexeme type (22) not found ...`

A more helpful error message could be:
```c
"Couldn't parse parameter override rule: '%s'. Error: provided parameter argument is missing ':=' followed by its value", argv[i + 1]);
```
Would this change be acceptable? I don't think `prev_error_string` adds much value here, but I could be wrong and it is more relevant for other types of errors.