ros2/rcl #1009: valgrind: Mismatched free / delete and other leaks

christianrauch, 2022-09-13:
## Bug report

valgrind reports a mismatched delete in `rcl_subscription_fini.part.0 (subscription.c:193)` and a couple of "definitely lost" leaks. Not sure if this originates in rclcpp pr rclc. See full valgrind log below.

**Required Info:**

- Operating System: Ubuntu 22.04
- Installation type: binaries
- Version or commit hash: humble
- DDS implementation: (default)
- Client library (if applicable): rclcpp 16.0.1 / rcl 5.3.2

#### Steps to reproduce issue
Run the `talker` with valgrind:
```
valgrind --leak-check=full /opt/ros/humble/lib/demo_nodes_cpp/talker
```

#### Expected behavior

No issues reported

#### Actual behavior

```
==155397== Memcheck, a memory error detector
==155397== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==155397== Using Valgrind-3.18.1 and LibVEX; rerun with -h for copyright info
==155397== Command: /opt/ros/humble/lib/demo_nodes_cpp/talker
==155397== 
[INFO] [1663107105.091217280] [talker]: Publishing: 'Hello World: 1'
[INFO] [1663107106.073742079] [talker]: Publishing: 'Hello World: 2'
[INFO] [1663107107.074266289] [talker]: Publishing: 'Hello World: 3'
^C[INFO] [1663107107.427783216] [rclcpp]: signal_handler(signum=2)
==155397== Mismatched free() / delete / delete []
==155397==    at 0x484BB6F: operator delete(void*, unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==155397==    by 0x4F205E5: rcl_subscription_fini.part.0 (subscription.c:193)
==155397==    by 0x49A2624: rclcpp::SubscriptionBase::SubscriptionBase(rclcpp::node_interfaces::NodeBaseInterface*, rosidl_message_type_support_t const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_subscription_options_s const&, bool)::{lambda(rcl_subscription_s*)#1}::operator()(rcl_subscription_s*) const (subscription_base.cpp:53)
==155397==    by 0x49A28E4: std::_Sp_counted_deleter<rcl_subscription_s*, rclcpp::SubscriptionBase::SubscriptionBase(rclcpp::node_interfaces::NodeBaseInterface*, rosidl_message_type_support_t const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_subscription_options_s const&, bool)::{lambda(rcl_subscription_s*)#1}, std::allocator<void>, (__gnu_cxx::_Lock_policy)2>::_M_dispose() (shared_ptr_base.h:442)
==155397==    by 0x492A079: std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() (shared_ptr_base.h:168)
==155397==    by 0x49A3814: rclcpp::SubscriptionBase::~SubscriptionBase() (shared_ptr_base.h:705)
==155397==    by 0x492A079: std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() (shared_ptr_base.h:168)
==155397==    by 0x49BCC3B: std::_Sp_counted_ptr_inplace<rclcpp::TimeSource::NodeState, std::allocator<rclcpp::TimeSource::NodeState>, (__gnu_cxx::_Lock_policy)2>::_M_dispose() (shared_ptr_base.h:705)
==155397==    by 0x492A079: std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() (shared_ptr_base.h:168)
==155397==    by 0x4970E1F: rclcpp::node_interfaces::NodeTimeSource::~NodeTimeSource() (node_time_source.cpp:53)
==155397==    by 0x49507E9: std::_Sp_counted_ptr<rclcpp::node_interfaces::NodeTimeSource*, (__gnu_cxx::_Lock_policy)2>::_M_dispose() (node_time_source.cpp:53)
==155397==    by 0x492A079: std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() (shared_ptr_base.h:168)
==155397==  Address 0xafa26c0 is 0 bytes inside a block of size 248 alloc'd
==155397==    at 0x484DA83: calloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==155397==    by 0x4F280B8: rcl_subscription_init (subscription.c:94)
==155397==    by 0x49A3357: rclcpp::SubscriptionBase::SubscriptionBase(rclcpp::node_interfaces::NodeBaseInterface*, rosidl_message_type_support_t const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_subscription_options_s const&, bool) (subscription_base.cpp:67)
==155397==    by 0x49B5F5D: std::_Function_handler<std::shared_ptr<rclcpp::SubscriptionBase> (rclcpp::node_interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&), rclcpp::create_subscription_factory<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, rclcpp::TimeSource::NodeState::attachNode(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeServicesInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeLoggingInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeClockInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeParametersInterface>)::{lambda(std::shared_ptr<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> > const>)#1}, std::allocator<void>, rclcpp::Subscription<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void>, rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, rclcpp::message_memory_strategy::MessageMemoryStrategy<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> > >, rclcpp::message_memory_strategy::MessageMemoryStrategy<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> >, rcl_interfaces::msg::ParameterEvent_<std::allocator<void> > >(rclcpp::TimeSource::NodeState::attachNode(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeServicesInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeLoggingInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeClockInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeParametersInterface>)::{lambda(std::shared_ptr<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> > const>)#1}&&, rclcpp::SubscriptionOptionsWithAllocator<std::allocator<void> > const&, rclcpp::message_memory_strategy::MessageMemoryStrategy<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> >::SharedPtr, std::shared_ptr<rclcpp::topic_statistics::SubscriptionTopicStatistics<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> > > >)::{lambda(rclcpp::node_interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&)#1}>::_M_invoke(std::_Any_data const&, rclcpp::node_interfaces::NodeBaseInterface*&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&) (subscription.hpp:147)
==155397==    by 0x49AC611: rclcpp::TimeSource::NodeState::attachNode(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeServicesInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeLoggingInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeClockInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeParametersInterface>) (std_function.h:590)
==155397==    by 0x49ADD91: rclcpp::TimeSource::attachNode(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeServicesInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeLoggingInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeClockInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeParametersInterface>) (time_source.cpp:507)
==155397==    by 0x49710FA: rclcpp::node_interfaces::NodeTimeSource::NodeTimeSource(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeServicesInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeLoggingInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeClockInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeParametersInterface>, rclcpp::QoS const&, bool) (node_time_source.cpp:41)
[... cut ...]
==155397==      possibly lost: 0 bytes in 0 blocks
==155397==    still reachable: 44,435 bytes in 149 blocks
==155397==         suppressed: 0 bytes in 0 blocks
==155397== Reachable blocks (those to which a pointer was found) are not shown.
==155397== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==155397== 
==155397== For lists of detected and suppressed errors, rerun with: -s
==155397== ERROR SUMMARY: 11 errors from 11 contexts (suppressed: 0 from 0)
```

#### Additional information

fujitatomoya, 2022-09-13:
I think https://github.com/ros2/rclcpp/pull/1995 can fix this problem, duplicated with https://github.com/ros2/rclcpp/issues/1996

clalancette, 2022-09-14:
> I think [ros2/rclcpp#1995](https://github.com/ros2/rclcpp/pull/1995) can fix this problem, duplicated with [ros2/rclcpp#1996](https://github.com/ros2/rclcpp/issues/1996)

Yes, confirmed that ros2/rclcpp#1995 fixes the `Mismatched free` problem.  The other leaks are less clear to me, they'll have to be investigated separately.

fujitatomoya, 2022-09-15:
> confirmed that https://github.com/ros2/rclcpp/pull/1995 fixes the Mismatched free problem.

this has been addressed.

everything else, with my local environment.

- 1 of 9 is `rmw_fastrtps` related.
- 8 of 9 are `class_loader` related.

<details><summary>valgrind --leak-check=full ./build/demo_nodes_cpp/talker</summary><div>

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# valgrind --leak-check=full ./build/demo_nodes_cpp/talker
==207805== Memcheck, a memory error detector
==207805== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==207805== Using Valgrind-3.18.1 and LibVEX; rerun with -h for copyright info
==207805== Command: ./build/demo_nodes_cpp/talker
==207805==
==207805== Syscall param ioctl(SIOCGIFHWADDR) points to uninitialised byte(s)
==207805==    at 0x4F21AFF: ioctl (ioctl.c:36)
==207805==    by 0x5F984EE: eprosima::fastrtps::rtps::IPFinder::getAllMACAddress(std::vector<eprosima::fastrtps::rtps::IPFinder::info_MAC, std::allocator<eprosima::fastrtps::rtps::IPFinder::info_MAC> >*) (in /root/ros2_ws/colcon_ws/install/fastrtps/lib/libfastrtps.so.2.8.0)
==207805==    by 0x5CAEE29: eprosima::fastdds::rtps::GuidUtils::GuidUtils() (in /root/ros2_ws/colcon_ws/install/fastrtps/lib/libfastrtps.so.2.8.0)
==207805==    by 0x5CA951A: eprosima::fastrtps::rtps::RTPSDomainImpl::create_participant_guid(int&, eprosima::fastrtps::rtps::GUID_t&) (in /root/ros2_ws/colcon_ws/install/fastrtps/lib/libfastrtps.so.2.8.0)
==207805==    by 0x5D64747: eprosima::fastdds::dds::DomainParticipantImpl::DomainParticipantImpl(eprosima::fastdds::dds::DomainParticipant*, unsigned int, eprosima::fastdds::dds::DomainParticipantQos const&, eprosima::fastdds::dds::DomainParticipantListener*) (in /root/ros2_ws/colcon_ws/install/fastrtps/lib/libfastrtps.so.2.8.0)
==207805==    by 0x5D4CC7E: eprosima::fastdds::dds::DomainParticipantFactory::create_participant(unsigned int, eprosima::fastdds::dds::DomainParticipantQos const&, eprosima::fastdds::dds::DomainParticipantListener*, eprosima::fastdds::dds::StatusMask const&) (in /root/ros2_ws/colcon_ws/install/fastrtps/lib/libfastrtps.so.2.8.0)
==207805==    by 0x58A9729: __create_participant(char const*, eprosima::fastdds::dds::DomainParticipantQos const&, bool, publishing_mode_t, rmw_dds_common::Context*, unsigned long) (in /root/ros2_ws/colcon_ws/install/rmw_fastrtps_shared_cpp/lib/librmw_fastrtps_shared_cpp.so)
==207805==    by 0x58AA282: rmw_fastrtps_shared_cpp::create_participant(char const*, unsigned long, rmw_security_options_s const*, bool, char const*, rmw_dds_common::Context*) (in /root/ros2_ws/colcon_ws/install/rmw_fastrtps_shared_cpp/lib/librmw_fastrtps_shared_cpp.so)
==207805==    by 0x57499BF: init_context_impl(rmw_context_s*) (in /root/ros2_ws/colcon_ws/build/rmw_fastrtps_cpp/librmw_fastrtps_cpp.so)
==207805==    by 0x574A100: rmw_fastrtps_cpp::increment_context_impl_ref_count(rmw_context_s*) (in /root/ros2_ws/colcon_ws/build/rmw_fastrtps_cpp/librmw_fastrtps_cpp.so)
==207805==    by 0x576FB0B: rmw_create_node (in /root/ros2_ws/colcon_ws/build/rmw_fastrtps_cpp/librmw_fastrtps_cpp.so)
==207805==    by 0x505C0C5: rcl_node_init (in /root/ros2_ws/colcon_ws/install/rcl/lib/librcl.so)
==207805==  Address 0x1ffefdff5f is on thread 1's stack
==207805==  in frame #1, created by eprosima::fastrtps::rtps::IPFinder::getAllMACAddress(std::vector<eprosima::fastrtps::rtps::IPFinder::info_MAC, std::allocator<eprosima::fastrtps::rtps::IPFinder::info_MAC> >*) (???:)
==207805==
[INFO] [1663255182.400603270] [talker]: Publishing: 'Hello World: 1'
[INFO] [1663255183.391459588] [talker]: Publishing: 'Hello World: 2'
[INFO] [1663255184.391885194] [talker]: Publishing: 'Hello World: 3'
[INFO] [1663255185.391716617] [talker]: Publishing: 'Hello World: 4'
[INFO] [1663255186.391714986] [talker]: Publishing: 'Hello World: 5'
^C[INFO] [1663255186.551842763] [rclcpp]: signal_handler(signum=2)
==207805==
==207805== HEAP SUMMARY:
==207805==     in use at exit: 53,205 bytes in 224 blocks
==207805==   total heap usage: 22,554 allocs, 22,330 frees, 5,755,981 bytes allocated
==207805==
==207805== 336 (16 direct, 320 indirect) bytes in 1 blocks are definitely lost in loss record 109 of 125
==207805==    at 0x484F013: operator new(unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==207805==    by 0x710E4E9: void class_loader::impl::registerPlugin<rclcpp_components::NodeFactoryTemplate<demo_nodes_cpp::Talker>, rclcpp_components::NodeFactory>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /root/ros2_ws/colcon_ws/build/demo_nodes_cpp/libtopics_library.so)
==207805==    by 0x70D7157: _GLOBAL__sub_I_talker.cpp (in /root/ros2_ws/colcon_ws/build/demo_nodes_cpp/libtopics_library.so)
==207805==    by 0x400647D: call_init.part.0 (dl-init.c:70)
==207805==    by 0x4006567: call_init (dl-init.c:33)
==207805==    by 0x4006567: _dl_init (dl-init.c:117)
==207805==    by 0x4F7BC84: _dl_catch_exception (dl-error-skeleton.c:182)
==207805==    by 0x400DFF5: dl_open_worker (dl-open.c:808)
==207805==    by 0x400DFF5: dl_open_worker (dl-open.c:771)
==207805==    by 0x4F7BC27: _dl_catch_exception (dl-error-skeleton.c:208)
==207805==    by 0x400E34D: _dl_open (dl-open.c:883)
==207805==    by 0x4E976BB: dlopen_doit (dlopen.c:56)
==207805==    by 0x4F7BC27: _dl_catch_exception (dl-error-skeleton.c:208)
[... cut ...]
==207805==    still reachable: 50,250 bytes in 166 blocks
==207805==         suppressed: 0 bytes in 0 blocks
==207805== Reachable blocks (those to which a pointer was found) are not shown.
==207805== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==207805==
==207805== Use --track-origins=yes to see where uninitialised values come from
==207805== For lists of detected and suppressed errors, rerun with: -s
==207805== ERROR SUMMARY: 9 errors from 9 contexts (suppressed: 0 from 0)
```

</div></details>

clalancette, 2022-09-15:
> * 8 of 9 are `class_loader` related.

This might be solved by https://github.com/ros/class_loader/pull/199

fujitatomoya, 2022-09-15:
> 1 of 9 is rmw_fastrtps related.

https://github.com/eProsima/Fast-DDS/pull/2954 can address above.

iuhilnehc-ynos, 2022-09-16:
Actually, I think https://github.com/ros/class_loader/pull/196 is the correct fix.

https://github.com/ros/class_loader/pull/199 update the following source code seems not good to me.
```patch
-   * @brief Destructor for the class. THIS MUST NOT BE VIRTUAL AND OVERRIDDEN BY
-   * TEMPLATE SUBCLASSES, OTHERWISE THEY WILL PULL IN A REDUNDANT METAOBJECT
-   * DESTRUCTOR OUTSIDE OF libclass_loader WITHIN THE PLUGIN LIBRARY! T
+   * @brief Default virtual destructor
   */
-  ~AbstractMetaObjectBase();
+  virtual ~AbstractMetaObjectBase() = default;
```

Using the https://github.com/ros/class_loader/pull/196 and the following patch for rclcpp_component, the memory leak will be gone.

```patch
diff --git a/rclcpp_components/src/node_main.cpp.in b/rclcpp_components/src/node_main.cpp.in
index 0ca5eb8c..71754d1f 100644
--- a/rclcpp_components/src/node_main.cpp.in
+++ b/rclcpp_components/src/node_main.cpp.in
@@ -30,14 +30,13 @@ int main(int argc, char * argv[])
   rclcpp::executors::@executor@ exec;
   rclcpp::NodeOptions options;
   options.arguments(args);
-  std::vector<class_loader::ClassLoader * > loaders;
   std::vector<rclcpp_components::NodeInstanceWrapper> node_wrappers;
 
   std::string library_name = "@library_name@";
   std::string class_name = "rclcpp_components::NodeFactoryTemplate<@component@>";
 
   RCLCPP_DEBUG(logger, "Load library %s", library_name.c_str());
-  auto loader = new class_loader::ClassLoader(library_name);
+  auto loader = std::make_unique<class_loader::ClassLoader>(library_name);
   auto classes = loader->getAvailableClasses<rclcpp_components::NodeFactory>();
   for (const auto & clazz : classes) {
     std::string name = clazz.c_str();
@@ -59,8 +58,6 @@ int main(int argc, char * argv[])
       exec.add_node(node);
     }
   }
-  loaders.push_back(loader);
-
 
   exec.spin();

```

<details><summary> log </summary>
<p>

```shell
chenlh ros2-master $ valgrind --leak-check=full ./build/demo_nodes_cpp/talker
==1213095== Memcheck, a memory error detector
==1213095== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==1213095== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==1213095== Command: ./build/demo_nodes_cpp/talker
==1213095== 
[INFO] [1663306906.483925290] [talker]: Publishing: 'Hello World: 1'
[INFO] [1663306907.465399901] [talker]: Publishing: 'Hello World: 2'
[INFO] [1663306908.465337444] [talker]: Publishing: 'Hello World: 3'
^C[INFO] [1663306908.918579342] [rclcpp]: signal_handler(signum=2)
==1213095== 
==1213095== HEAP SUMMARY:
==1213095==     in use at exit: 49,916 bytes in 167 blocks
==1213095==   total heap usage: 21,331 allocs, 21,164 frees, 5,362,526 bytes allocated
==1213095== 
==1213095== LEAK SUMMARY:
==1213095==    definitely lost: 0 bytes in 0 blocks
==1213095==    indirectly lost: 0 bytes in 0 blocks
==1213095==      possibly lost: 0 bytes in 0 blocks
==1213095==    still reachable: 49,916 bytes in 167 blocks
==1213095==         suppressed: 0 bytes in 0 blocks
==1213095== Reachable blocks (those to which a pointer was found) are not shown.
==1213095== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==1213095== 
==1213095== For lists of detected and suppressed errors, rerun with: -s
==1213095== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)

```

</p>
</details>

tylerjw, 2022-10-21:
> updating the following source code seems not good to me.

What does "seems not good to me" mean? Can you give a technical argument?

clalancette, 2026-11-04:
I think most of this was fixed either in ros2/rclcpp#1995, or in the other packages.  I'm going to close this out for now (especially given the age), but if you see more leaks please feel free to report them.

closed 2026-11-04 by clalancette