ros2/rcl #1026: yaml tag full support in rcl_yaml_param_parser

fujitatomoya, 2023-01-04:
## Feature request

#### Feature description

discussion https://github.com/ros2/rcl/pull/999#issuecomment-1370387037, it would be better to support yaml tag full support, not only with `!!str`.

#### Implementation considerations

Reference implementation can be https://github.com/ros2/rcl/pull/999 which added `!!str` tag.

llapx, 2023-01-12:
@fujitatomoya 

yaml tag [timestamp](https://yaml.org/type/timestamp.html), need a regex check, which need compiling in a regex related library, and ros2 parameter type does not support timestamp, do we really need it?

llapx, 2023-02-02:
#1032

clalancette, 2026-11-04:
This was implemented a long time ago, so closing this out.

closed 2026-11-04 by clalancette