ros2/rcl #1034: Anonymous Nodes

danthony06, 2023-02-22:
## Feature request

#### Feature description
Enable anonymous node names.

#### Implementation considerations
We frequently used anonymous nodes in ROS 1 when developing debugging and introspection tools. I see a lot of references to anonymous nodes throughout the core ROS2 libraries, and inside rcl/node_options.h, but it seems to have been disabled or abandoned. I also saw this comment mentioning that they're a desirable feature: https://github.com/ros2/rcl/issues/375#issuecomment-458648680. Is there any reason they have not been implemented in ROS 2?

I think ROS1 just used the wall timer to generate a unique ID. Will this approach work in ROS 2?

fujitatomoya, 2023-02-24:
This node option was considered long time ago, but never implemented yet.

https://github.com/ros2/rcl/blob/2d34ba831fc4693e2e44af2da40281f393ed88c8/rcl/include/rcl/node_options.h#L36

I think that this node option is useful for user application.

danthony06, 2023-02-24:
I've started working on adding this, and have a partial implementation.

methylDragon, 2023-03-02:
This would be great to have!

Note that any implementation of this should append the anonymisation suffix **after** any node name and node namespace remaps are done (as well as after parameters are loaded.)

Another useful bonus would be if a CLI argument could be used to configure this behavior too.

danthony06, 2023-03-02:
Thanks, I was planning on appending a random number to a given node name,
if the user supplies it. In ROS 1 we used the wall time as the anonymous
node name. Does this still sound like a good approach?

methylDragon, 2023-03-02:
I think following ROS 1 would be okay, and it has the added benefit of giving that little bit more information that helps to order nodes (alphanumerically, I mean)