ros2/rcl #1036: Custom logging allocator not passed to rcutils_logging_initialize_with_allocator

jrutgeer, 2023-03-01:
Logging is initialized by a call to `rcl_logging_configure_with_output_handler()` (e.g. from `rclcpp::init()`).

This calls `RCUTILS_LOGGING_AUTOINIT` and then sets the logger allocator:
https://github.com/ros2/rcl/blob/4eccc3cbe65f5b58981f22efaf612a65731cb2f0/rcl/src/rcl/logging.c#L65-L66

I think this is a bug, as `RCUTILS_LOGGING_AUTOINIT` resolves to `rcutils_logging_initialize()` which calls `rcutils_logging_initialize_with_allocator(rcutils_get_default_allocator());`, i.e. with the default allocator (malloc) instead of the provided one. So the logging hash table gets the default allocator and not the provided one.

https://github.com/ros2/rcutils/blob/61018b2f88e55ac81edda4a45a02634493c999ed/include/rcutils/logging.h#L560-L572

https://github.com/ros2/rcutils/blob/61018b2f88e55ac81edda4a45a02634493c999ed/src/logging.c#L129-L132

jrutgeer, 2023-03-16:
Haven't had a chance to look at it in detail, but on first sight I don't think this will fix it as each log call calls RCUTILS_LOGGING_AUTOINIT and there are log calls before `rcl_logging_configure_with_output_handler()`

fujitatomoya, 2023-03-17:
Ah yeah, that is true, i was focusing on rclcpp. but if application calls rcl or rcutils directly to initialize the logger, that is gonna be the problem.

> See also https://github.com/ros2/rcl/issues/1037#issuecomment-1462933219

thanks, i will take a look at it.

fujitatomoya, 2023-03-21:
@jrutgeer as you mentioned https://github.com/ros2/rcl/issues/1036#issuecomment-1472885013, this is related to https://github.com/ros2/rcl/issues/1037. So i will keep that on https://github.com/ros2/rcl/issues/1037 instead, closing this issue since `[Custom logging allocator not passed to rcutils_logging_initialize_with_allocator](https://github.com/ros2/rcl/issues/1036#top)` is addressed.

closed 2023-03-21 by fujitatomoya