ros2/rcl #1040: Rosout publisher always gets standard allocator

jrutgeer, 2023-03-05:
`rcl_logging_rosout_init_publisher_for_node()` calls `rcl_publisher_init()` with default publisher options:
https://github.com/ros2/rcl/blob/d15594effa63065a19a9f69960ea80f5ac5be8bd/rcl/src/rcl/logging_rosout.c#L275

https://github.com/ros2/rcl/blob/d15594effa63065a19a9f69960ea80f5ac5be8bd/rcl/src/rcl/logging_rosout.c#L266

So this always uses the standard allocator, even if a custom allocator is provided through the node options.

fujitatomoya, 2023-03-09:
@jrutgeer thanks for creating issue. Good eye 👍 , this is the problem for real-time system cz that will call dynamic memory allocation even if user specifies custom allocator.

fujitatomoya, 2023-03-09:
@jrutgeer if you have time, can you review https://github.com/ros2/rcl/pull/1044, that will be helpful.

fujitatomoya, 2023-03-16:
closing this, backport to humble is also completed. https://github.com/ros2/rcl/pull/1045

closed 2023-03-16 by fujitatomoya