ros2/rcl #1041: Contradicting code docs for rcl_publish and rmw_publish

jrutgeer, 2023-03-06:
rcl_publish states the following:

https://github.com/ros2/rcl/blob/60d80648c2a7cdbadcc7f22bdbc3c358c04dff7c/rcl/include/rcl/publisher.h#L303-L305

However the `rmw_publish()` doc seem to contradict this for the asynchronous case:

https://github.com/ros2/rmw/blob/555bb905e8f52dc0a0f43361a0b3f19b474ec99e/rmw/include/rmw/rmw.h#L509-L514

```
 * \par Runtime behavior
 *   It is implementation defined whether to publish a ROS message is a
 *   synchronous or asynchronous, blocking or non-blocking operation.
 *   However, asynchronous implementations are not allowed to access the
 *   given ROS message after this function returns.
 *   Check the implementation documentation to learn about publish behavior.
 *
```

I assume the rmw doc is the correct one?

clalancette, 2023-03-06:
> I assume the rmw doc is the correct one?

Yes, correct.  If you'd like to submit a PR to update the rcl one, we'd be happy to review.

CursedRock17, 2023-09-24:
The fix to this issue seems to have been merged, so this can probably be closed.

closed 2023-09-25 by jrutgeer