ros2/rcl #1061: master branch is behind rolling branch

pablogs9, 2023-04-12:
## Bug report

`master` to `rolling` mirror action is failing: https://github.com/ros2/rcl/actions/workflows/mirror-rolling-to-master.yaml

For some legacy packages in micro-ROS we rely on `master` instead of `rolling`, we have not migrated yet the whole set of packages and they can fail.

**Required Info:**

N/A

#### Steps to reproduce issue

N/A

#### Expected behavior

`master` is mirrored to `rolling`

#### Actual behavior

`master` is some commit behind.

#### Additional information

N/A

iuhilnehc-ynos, 2023-04-12:
https://github.com/ros2/rcl/pull/1053 used the wrong branch (master) to push, so the github action can't push the rolling into master.

clalancette, 2023-04-12:
Honestly, we are just going to get rid of this syncing.  It is causing more problems than it is solving, and it has been a whole year since we made the change.  I don't have time to do it right now, but @pablogs9 you should update microROS to use the `rolling` branch instead of `master`.

closed 2023-04-12 by pablogs9