ros2/rcl #1068: Fix durability of service introspection topics

nirwester, 2023-05-11:
Hi, while testing the new service introspection topic, I noticed an anomaly on the durability of the published service response.

If executing the following sequence of actions:

-starting a server
-starting a client
-wait some seconds to make sure that the execution is finished
-subscribe to the introspection topic (ex ros2 topic echo /add_two_ints/_service_event)
The last response sent by the server is received, even if the service is no more running. Ex:

info:
event_type: 2
stamp:
sec: 1683653877
nanosec: 751816926
client_gid:
...
sequence_number: 1
request: []
response:
sum: 3

I used this code, where the basic services example was modified to enable full intropsection:
https://github.com/nirwester/ros2_journey_examples/tree/master/homeworks/chap4/service_with_introspection
Running on this docker container (jammy-Iron):
https://github.com/osrf/docker_images/pull/673

This would seem an issue in the durability of the corresponding publisher, probably set to transient_local. I don't know if there's any technical reason for this kind of implementation, but getting the last published results upon subscription seems confusing to me.

The problem was reproduced by @clalancette that proposed two possible solutions:

- Leave it as TRANSIENT_LOCAL, but give it a large queue depth. That means that late-joining subscribers would be able to get all of the previous data, at the expense of using a lot of memory.
- Change it to VOLATILE and just don't worry about late-joining subscribers.

The volatile choice sounds like the more reasonable, but we might be missing something.

jacobperron, 2023-05-11:
Volatile durability makes sense to me. I believe this is what is suggested in the design doc: https://github.com/ros-infrastructure/rep/pull/360

> The service event topics proposed in this REP shall use the default quality of service settings.

I'm pretty sure that the default durability QoS setting for topics is volatile. If someone wants to change the QoS for a specific service  event topic, they can still do that with a runtime config file.

fujitatomoya, 2023-05-12:
probably I might be mistaken...

currently `configure_introspection` allows user application to set QoS for service event publisher. that said user can change the QoS based on the use case, that leaves the flexibility for the user application. (example sets `SystemDefaultsQoS` that leads to the implementation specific setting)

why do we want to set a static QoS for service event publisher to remove the flexibility for user application?

nirwester, 2023-05-12:
@fujitatomoya It apparently can, yet doing something like:

```
service->configure_introspection(node->get_clock(),
                                   rclcpp::SystemDefaultsQoS(),
                                   RCL_SERVICE_INTROSPECTION_CONTENTS);
```
                                   
Results in a transient local durability on the published result, which looks strange to me.

nirwester, 2023-05-12:
For completeness, here are the details of the published topic when using SystemDefaultQoS on both service and client:
```
ros2 topic info -v /add_two_ints/_service_event
Type: example_interfaces/srv/AddTwoInts_Event

Publisher count: 1

Node name: add_two_ints_server
Node namespace: /
Topic type: example_interfaces/srv/AddTwoInts_Event
Topic type hash: UNKNOWN
Endpoint type: PUBLISHER
GID: 01.0f.a2.5c.c5.00.59.e0.00.00.00.00.00.00.14.03
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: TRANSIENT_LOCAL
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 0
```

fujitatomoya, 2023-05-12:
@nirwester yes i understand, and that because application sets `SystemDefaultQoS`, right?

i mean, can user not do the following? that explicitly sets `RMW_QOS_POLICY_DURABILITY_VOLATILE`? (just one of example QoS setting to set `RMW_QOS_POLICY_DURABILITY_VOLATILE`, user actually have full control for any QoS setting.)

```cpp
service->configure_introspection(node->get_clock(),
                                   rclcpp::ServicesQoS(),
                                   RCL_SERVICE_INTROSPECTION_CONTENTS);
```

https://github.com/ros2/rmw/blob/36003aaed8fd9718c20b19f4da6fb21929574058/rmw/include/rmw/qos_profiles.h#LL64C1-L75C3

nirwester, 2023-05-12:
Understood, thanks a lot for the clear explanation! Given these differences, that can deeply influence the behavior of the application, I'll stay away from rclcpp::SystemDefaultsQoS :)

closed 2023-05-12 by nirwester

fujitatomoya, 2023-05-12:
@iuhilnehc-ynos thanks for the follow-up.

@nirwester if you find any other issues, please feel free to open it.

btw, i was thinking that probably we can set the default QoS with durability `VOLATILE` via service event publisher, if user does not specify the QoS for that? i think that what happened here, user probably sets `SystemDefaultsQoS` and that leads to this question.