ros2/rcl #1162: :farmer: Flaky `test_action_interaction__rmw_connextdds` in Windows debug

Crola1702, 2024-06-26:
## Bug report

**Required Info:**

- Operating System:
  - Windows
- Installation type:
  - Source
- Version or commit hash:
  - Rolling
- DDS implementation:
  - ConnextDDS

#### Steps to reproduce issue
1. Run a build in [nightly_win_deb](https://ci.ros2.org/view/nightly/job/nightly_win_deb/)
2. See test regressions fail

#### Additional information

Reference build: https://ci.ros2.org/view/nightly/job/nightly_win_deb/3136/

Test regressions:
* [projectroot.test_action_interaction__rmw_connextdds](https://ci.ros2.org/view/nightly/job/nightly_win_deb/3136/testReport/(root)/projectroot/test_action_interaction__rmw_connextdds/)
* [rcl_action.TestActionClientServerInteraction__rmw_connextdds.test_interaction](https://ci.ros2.org/view/nightly/job/nightly_win_deb/3136/testReport/rcl_action/TestActionClientServerInteraction__rmw_connextdds/test_interaction/)
* [rcl_action.TestActionClientServerInteraction__rmw_connextdds.test_interaction_with_cancel](https://ci.ros2.org/view/nightly/job/nightly_win_deb/3136/testReport/rcl_action/TestActionClientServerInteraction__rmw_connextdds/test_interaction_with_cancel/)

<details>
<summary>
Log output:
</summary>

```
-- run_test.py: extra environment variables:
 - RCL_ASSERT_RMW_ID_MATCHES=rmw_connextdds
 - RMW_IMPLEMENTATION=rmw_connextdds
-- run_test.py: invoking following command in 'C:\ci\ws\build\rcl_action':
 - C:/ci/ws/build/rcl_action/Debug/test_action_interaction__rmw_connextdds.exe --gtest_output=xml:C:/ci/ws/build/rcl_action/test_results/rcl_action/test_action_interaction__rmw_connextdds.gtest.xml
Running main() from C:\ci\ws\install\src\gtest_vendor\src\gtest_main.cc
[==========] Running 2 tests from 1 test suite.
[----------] Global test environment set-up.
[----------] 2 tests from TestActionClientServerInteraction__rmw_connextdds
[ RUN      ] TestActionClientServerInteraction__rmw_connextdds.test_interaction
C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(227): error: Expected equality of these values:
  ret
    Which is: 2
  0
error not set

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(238): error: Value of: this->is_goal_request_ready
  Actual: false
Expected: true
error not set

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(246): error: Expected equality of these values:
  ret
    Which is: 2201
  0
error not set

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(249): error: Expected equality of these values:
  this->outgoing_goal_request.goal.order
    Which is: 10
  this->incoming_goal_request.goal.order
    Which is: -842150451

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(253): error: Value of: (0 == memcmp(this->outgoing_goal_request.goal_id.uuid, this->incoming_goal_request.goal_id.uuid, 16))
  Actual: false
Expected: true

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(274): error: Expected equality of these values:
  ret
    Which is: 2
  0
error not set

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(287): error: Value of: this->is_goal_response_ready
  Actual: false
Expected: true

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(296): error: Expected equality of these values:
  ret
    Which is: 2103
  0
error not set

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(299): error: Expected equality of these values:
  this->outgoing_goal_response.accepted
    Which is: true
  this->incoming_goal_response.accepted
    Which is: true

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(300): error: Expected equality of these values:
  this->outgoing_goal_response.stamp.sec
    Which is: 123
  this->incoming_goal_response.stamp.sec
    Which is: -842150451

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(301): error: Expected equality of these values:
  this->outgoing_goal_response.stamp.nanosec
    Which is: 456789
  this->incoming_goal_response.stamp.nanosec
    Which is: 3452816845

[  FAILED  ] TestActionClientServerInteraction__rmw_connextdds.test_interaction (2548 ms)
[ RUN      ] TestActionClientServerInteraction__rmw_connextdds.test_interaction_with_cancel
C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(500): error: Expected equality of these values:
  ret
    Which is: 2
  0
error not set

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(511): error: Value of: this->is_goal_request_ready
  Actual: false
Expected: true
error not set

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(519): error: Expected equality of these values:
  ret
    Which is: 2201
  0
error not set

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(522): error: Expected equality of these values:
  this->outgoing_goal_request.goal.order
    Which is: 10
  this->incoming_goal_request.goal.order
    Which is: -842150451

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(526): error: Value of: (0 == memcmp(this->outgoing_goal_request.goal_id.uuid, this->incoming_goal_request.goal_id.uuid, 16))
  Actual: false
Expected: true

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(547): error: Expected equality of these values:
  ret
    Which is: 2
  0
error not set

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(560): error: Value of: this->is_goal_response_ready
  Actual: false
Expected: true

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(569): error: Expected equality of these values:
  ret
    Which is: 2103
  0
error not set

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(572): error: Expected equality of these values:
  this->outgoing_goal_response.accepted
    Which is: true
  this->incoming_goal_response.accepted
    Which is: true

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(573): error: Expected equality of these values:
  this->outgoing_goal_response.stamp.sec
    Which is: 123
  this->incoming_goal_response.stamp.sec
    Which is: -842150451

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(574): error: Expected equality of these values:
  this->outgoing_goal_response.stamp.nanosec
    Which is: 456789
  this->incoming_goal_response.stamp.nanosec
    Which is: 3452816845

[  FAILED  ] TestActionClientServerInteraction__rmw_connextdds.test_interaction_with_cancel (2414 ms)
[----------] 2 tests from TestActionClientServerInteraction__rmw_connextdds (4964 ms total)

[----------] Global test environment tear-down
[==========] 2 tests from 1 test suite ran. (4964 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 2 tests, listed below:
[  FAILED  ] TestActionClientServerInteraction__rmw_connextdds.test_interaction
[  FAILED  ] TestActionClientServerInteraction__rmw_connextdds.test_interaction_with_cancel

 2 FAILED TESTS
-- run_test.py: return code 1
-- run_test.py: inject classname prefix into gtest result file 'C:/ci/ws/build/rcl_action/test_results/rcl_action/test_action_interaction__rmw_connextdds.gtest.xml'
-- run_test.py: verify result file 'C:/ci/ws/build/rcl_action/test_results/rcl_action/test_action_interaction__rmw_connextdds.gtest.xml'
```

</details>

Flaky report (last 15 days):

job_name|last_fail|first_fail|build_count|failure_count|failure_percentage
-- | -- | -- | -- | -- | --
nightly_win_deb|2024-06-26|2024-06-12|14|4|28.57
nightly_win_rel|2024-06-25|2024-06-25|16|1|6.25

Crola1702, 2024-07-05:
The flaky ratio of these tests has grown up considerably:

job_name|last_fail|first_fail|build_count|failure_count|failure_percentage
-- | -- | -- | -- | -- | --
nightly_win_rel|2024-07-05|2024-06-25|16|8|50.0
nightly_win_deb|2024-07-05|2024-06-21|12|9|75.0

Crola1702, 2024-07-29:
This is still hapenning in Windows Debug: [nightly_win_deb#3169](https://ci.ros2.org/view/nightly/job/nightly_win_deb/3169). Test failure: https://ci.ros2.org/view/nightly/job/nightly_win_deb/3169/testReport/(root)/projectroot/test_action_interaction__rmw_connextdds/

Crola1702, 2024-08-12:
These tests are failing consistently in Windows Debug:
* [rcl_action.TestActionClientServerInteraction.test_interaction](https://ci.ros2.org/job/nightly_win_deb/3183/testReport/junit/rcl_action/TestActionClientServerInteraction/test_interaction/)
* [rcl_action.TestActionClientServerInteraction.test_interaction_with_cancel](https://ci.ros2.org/job/nightly_win_deb/3183/testReport/junit/rcl_action/TestActionClientServerInteraction/test_interaction_with_cancel/)
* [projectroot.test_action_interaction__rmw_connextdds](https://ci.ros2.org/job/nightly_win_deb/3183/testReport/junit/(root)/projectroot/test_action_interaction__rmw_connextdds/)

<details>
<summary>
Log output:
</summary>

```
-- run_test.py: extra environment variables:
 - RMW_IMPLEMENTATION=rmw_connextdds
-- run_test.py: invoking following command in 'C:\ci\ws\build\rcl_action':
 - C:/ci/ws/build/rcl_action/Debug/test_action_interaction.exe --gtest_output=xml:C:/ci/ws/build/rcl_action/test_results/rcl_action/test_action_interaction__rmw_connextdds.gtest.xml
Running main() from C:\ci\ws\install\src\gtest_vendor\src\gtest_main.cc
[==========] Running 2 tests from 1 test suite.
[----------] Global test environment set-up.
[----------] 2 tests from TestActionClientServerInteraction
[ RUN      ] TestActionClientServerInteraction.test_interaction
C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(222): error: Expected equality of these values:
  ret
    Which is: 2
  0
error not set

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(233): error: Value of: this->is_goal_request_ready
  Actual: false
Expected: true
error not set

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(241): error: Expected equality of these values:
  ret
    Which is: 2201
  0
error not set

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(244): error: Expected equality of these values:
  this->outgoing_goal_request.goal.order
    Which is: 10
  this->incoming_goal_request.goal.order
    Which is: -842150451

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(248): error: Value of: (0 == memcmp(this->outgoing_goal_request.goal_id.uuid, this->incoming_goal_request.goal_id.uuid, 16))
  Actual: false
Expected: true

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(269): error: Expected equality of these values:
  ret
    Which is: 2
  0
error not set

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(282): error: Value of: this->is_goal_response_ready
  Actual: false
Expected: true

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(291): error: Expected equality of these values:
  ret
    Which is: 2103
  0
error not set

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(294): error: Expected equality of these values:
  this->outgoing_goal_response.accepted
    Which is: true
  this->incoming_goal_response.accepted
    Which is: true

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(295): error: Expected equality of these values:
[... 62 lines cut ...]
C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(567): error: Expected equality of these values:
  this->outgoing_goal_response.stamp.sec
    Which is: 123
  this->incoming_goal_response.stamp.sec
    Which is: -842150451

C:\ci\ws\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(568): error: Expected equality of these values:
  this->outgoing_goal_response.stamp.nanosec
    Which is: 456789
  this->incoming_goal_response.stamp.nanosec
    Which is: 3452816845

[  FAILED  ] TestActionClientServerInteraction.test_interaction_with_cancel (20526 ms)
[----------] 2 tests from TestActionClientServerInteraction (41094 ms total)

[----------] Global test environment tear-down
[==========] 2 tests from 1 test suite ran. (41094 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 2 tests, listed below:
[  FAILED  ] TestActionClientServerInteraction.test_interaction
```

</details>

CC @clalancette

MichaelOrlov, 2024-08-22:
@clalancette A friendly ping to follow up

miguelgonrod, 2026-10-26:
Surprissingly this keeps happening since 2024. 
job_name|last_fail|first_fail|build_count|failure_count|failure_percentage
|--|--|--|--|--|--|
nightly_win_rep|2026-10-26|2026-10-06|21|6|28.57
nightly_win_rel|2026-10-13|2026-09-26|23|5|21.74

Last failure https://ci.ros2.org/job/nightly_win_rep/4252

```
[ RUN      ] TestActionClientServerInteraction.test_interaction_with_cancel
W:\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(494): error: Expected equality of these values:
  ret
    Which is: 2
  0
error not set

W:\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(505): error: Value of: this->is_goal_request_ready
  Actual: false
Expected: true
error not set

W:\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(513): error: Expected equality of these values:
  ret
    Which is: 4201
  0
error not set

W:\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(516): error: Expected equality of these values:
  this->outgoing_goal_request.goal.order
    Which is: 10
  this->incoming_goal_request.goal.order
    Which is: 16777473

W:\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(520): error: Value of: (0 == memcmp(this->outgoing_goal_request.goal_id.uuid, this->incoming_goal_request.goal_id.uuid, 16))
  Actual: false
Expected: true

W:\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(541): error: Expected equality of these values:
  ret
    Which is: 2
  0
error not set

W:\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(554): error: Value of: this->is_goal_response_ready
  Actual: false
Expected: true

W:\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(563): error: Expected equality of these values:
  ret
    Which is: 4103
  0
error not set

W:\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(566): error: Expected equality of these values:
  this->outgoing_goal_response.accepted
    Which is: true
  this->incoming_goal_response.accepted
    Which is: false

W:\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(567): error: Expected equality of these values:
  this->outgoing_goal_response.stamp.sec
    Which is: 123
  this->incoming_goal_response.stamp.sec
    Which is: 0

W:\src\ros2\rcl\rcl_action\test\rcl_action\test_action_interaction.cpp(568): error: Expected equality of these values:
  this->outgoing_goal_response.stamp.nanosec
    Which is: 456789
  this->incoming_goal_response.stamp.nanosec
    Which is: 0

[  FAILED  ] TestActionClientServerInteraction.test_interaction_with_cancel (20242 ms)
```

@clalancette should we disable these tests? two years seems reasonable to conclude this tests aren't being tracked jaja

clalancette, 2026-10-26:
I'm doing a few other cleanups on tests now, let me see if I can track this one down.

clalancette, 2026-10-27:
I think #1354 will solve this, though we won't know for sure until it is merged and nightlies run for a few weeks.

PR #1172 (2024-07-25): Increase the test_action_interaction timeouts.
While I can't reproduce the problem locally, I suspect that waiting only 1 second for the entities to become ready isn't enough in all cases, particularly on Windows, with Connext, and when we are running in parallel with other tests. Thus, increase the timeout for the rcl_wait() in all of the test_action_interaction tests, which should hopefully be enough to make this always pass.

In theory this should fix #1162
closed 2026-10-27 by clalancette