ros2/rcl #1178: [FR] select the logger implementation without rebuilding

fujitatomoya, 2024-08-22:
## Feature request

#### Feature description

Currently there is no way to select the logging backend without building `rcl`.
This is because that when building `rcl`, we must choose the logging implementation to be used.

In default `rcl_logging_spdlog` is selected if `RCL_LOGGING_IMPLEMENTATION` is not set.
But we can change it with `RCL_LOGGING_IMPLEMENTATION` and available logging backend library.

e.g `rcl_logging_noop`

```bash
export RCL_LOGGING_IMPLEMENTATION=rcl_logging_noop
colcon build --symlink-install --cmake-clean-cache --packages-select rcl_logging_noop rcl
```

and, no log files are generated with `rcl_logging_noop`

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run demo_nodes_cpp talker
[INFO] [1724349615.489636579] [talker]: Publishing: 'Hello World: 1'
[INFO] [1724349616.489588912] [talker]: Publishing: 'Hello World: 2'
[INFO] [1724349617.489610598] [talker]: Publishing: 'Hello World: 3'
[INFO] [1724349618.489512697] [talker]: Publishing: 'Hello World: 4'
...
root@tomoyafujita:~/.ros/log# ls

```

This generates the constraint that whole system needs to use only specified logging backend in this workspace.
It would be nice if we can build the logger backend libraries, and select the logging backend when the program starts.

#### Implementation considerations

Current related code is,

https://github.com/ros2/rcl/blob/dcfe9ba4f95e3c0d546d30b3f03b769df16667a1/rcl/cmake/get_default_rcl_logging_implementation.cmake#L26-L34

Probably we can escalate `RCL_LOGGING_IMPLEMENTATION` like how `RMW_IMPLEMENTATION` works to select the rmw implementation when executing programs.

clalancette, 2024-08-23:
I'll state upfront that I'm not against this feature.

But are there any practical uses for it?  While I think that our logging subsystem has a number of problems, swapping out the disk logging backend at runtime doesn't seem to be one of them.  So I'd like to hear about a practical use before we spend time on it.

fujitatomoya, 2024-08-23:
@clalancette 

the use case (not production yet) that i have off the top of my head, to support 2 (or multiple) logger backend in the same system.

- one could be default `spdlog`, that just stores the logging files in specific file system. (on-system log just in case.)
- we have proprietary in-memory logger backend, some performance critical application uses. (once the memory chunk is full, squash those into the file system and send it to the cloud or something.)
- not all, but some nodes are system application, so that we want to send the log data via logging data pipeline (such as rsyslog, FluentBit) to entire system observability to collect and aggregate the log data, and give the feedback to the system.

note: i am spending a couple of days for `rcl_logging_syslog` (forwarded to FluentBit), so i can share more details about this. I think i can disclose the source code with some documentation in this weekend. (or hopefully today 😄 )

fujitatomoya, 2024-08-24:
@clalancette i developed some PoC based on `syslog` and support `FluentBit`, please see overview https://github.com/fujitatomoya/rcl_logging_syslog

fujitatomoya, 2025-02-07:
I am not sure if i can make this into Kilted, but had quick code scan to comprehend what needs to be done.

- introduce `rcl_logging_implementation` package in https://github.com/ros2/rcl_logging to abstract the logging implementation instead of [dynamically built with `rcl`](https://github.com/ros2/rcl/blob/1427bbb73af6bda69036e5efe1255dfb90329186/rcl/CMakeLists.txt#L23), so that `rcl` can only depend on `rcl_logging_implementation` and `rcl_logging_interfaces` without any implementation details.
- `rcl_logging_implementation` should be implemented similar with https://github.com/ros2/rmw_implementation, the main focus of this package is to resolve all the rcl logging interface symbols, support dynamic loading the user specified implementation and  connect symbols to them internally. 
- `rcl` probably should have `init` function in the process space to call `rcl_logging_indentifier_check` that makes sure it fails if user specifying unavailable logging implementation. (similarly https://github.com/ros2/rcl/blob/1427bbb73af6bda69036e5efe1255dfb90329186/rcl/src/rcl/rmw_implementation_identifier_check.c#L172-L178)
- `rcl` should always use default `rcl_logging_spdlog` if not specified `RCL_LOGGGING_IMPLEMENTATION` env variable at runtime.

I think we need a basic design document to describe more details before implementation.

clalancette, 2025-09-18:
> It would be the only way users could disable logs being written to disk while keeping their stdout logs without recompiling rcl (by switching it to the noop backend)

As I've mentioned elsewhere, that is not the case.  We already have `--disable-external-lib-logs` available.  While I understand you don't like the UI for that (and we can discuss that elsewhere), that use case doesn't require this feature in my opinion.

fujitatomoya, 2025-09-18:
besides https://github.com/ros2/rcl/issues/1178#issuecomment-2644237076,

in general, this decoupling should be much better.

- this allows user to plugin their own logging implementation without any source build of rcl. some users do not have their own CI/CD pipeline, in that case this can be burden.
- similar to above situation, we have own CI/CD pipeline to build the entire ros2 packages as base system. the logging implementation can be different based on the production use case, that says individual business units use different logging implementation. in this situation, we need to build the rcl package for each use cases. i do not think this is ideal.

Barry-Xu-2018, 2025-11-21:
> > The variable RCL_LOGGING_IMPLEMENTATION is used both at compile time and at runtime. Can we clearly separate its usage?
> 
> i think it describes the clear difference between them in google doc `RCL_LOGGING_IMPLEMENTATION environment variable` what is your unclear point on this?

Yes. The description is very clear. 
I'm just a bit concerned that using a variable at both compile time and runtime might cause confusion for users.

> > At runtime, a new environment variable LOGGING_IMPLEMENTATION is used to specify the final logging library to be used.
> 
> There will be no such environment varibale `LOGGING_IMPLEMENTATION`, but only `RCL_LOGGING_IMPLEMENTATION` as designed?

I just want to separate the variables used during compile time and runtime, which is why I proposed introducing this variable. Of course, this will result in the introduction of a new environment variable.  

This is just one of my thoughts. I don't have a strong inclination to make changes to the current design.

fujitatomoya, 2025-12-16:
@Barry-Xu-2018 just FYI, it is working with https://github.com/fujitatomoya/rcl_logging_syslog without any test failures.

fujitatomoya, 2025-12-16:
@Barry-Xu-2018 now i am requesting review and feedback from ROS PMC, let's wait for a while. after the PMC review, we can proceed to index/release this new `rcl_logging_implementation` package and documentation. thanks for your review!

fujitatomoya, 2026-01-16:
# rcl_logging_implementation Release Procedure

## Related PRs
- [[ros2/rcl_logging#135](https://github.com/ros2/rcl_logging/pull/135)](https://github.com/ros2/rcl_logging/pull/135) - Introduces `rcl_logging_implementation` package
- [[ros2/rcl#1276](https://github.com/ros2/rcl/pull/1276)](https://github.com/ros2/rcl/pull/1276) - Adds `rcl_logging_implementation` dependency to `rcl`

## Procedure

1. **Merge rcl_logging#135**
   - Introduces the new `rcl_logging_implementation` package
   - Safe to merge since nothing in mainline depends on it yet

2. **Bloom release `rcl_logging` repo**
   - Includes the new `rcl_logging_implementation` package

3. **Merge rcl#1276**
   - Adds dependency on `rcl_logging_implementation` in `rcl/package.xml`
   - Safe to merge now since `rcl_logging_implementation` is available in the distribution

4. **Bloom release `rcl` repo**
   - Release as normal

## Key Constraint

> **Do NOT merge rcl#1276** until `rcl_logging_implementation` is released and indexed in rosdistro.
>
> If rcl#1276 is merged before `rcl_logging_implementation` is available, CI and source builds will fail due to missing dependency.

fujitatomoya, 2026-01-18:
@Barry-Xu-2018 @Yadunund if you have time, please have a look at https://github.com/ros2/ros2_documentation/pull/6166?

peci1, 2026-01-18:
It's great you're moving forward with this :) Looking at the implementation, I think it should be possible to even switch the libraries multiple times during runtime, when I use `force_unload_logging_library()`, right? Not that I'd like to do it on a deployed code, but in tests it could be super helpful. That way, it would be possible to test e.g. that a function actually logs something, and to capture that something in a variable without it being written to the disk - but later switching back to the default logger so that other tests can utilize normal logging behavior. Am I right? Is this the intended behavior?

fujitatomoya, 2026-01-18:
yeah, we can do that. but `force_unload_logging_library` is meant to be only for test because i do not want to introduce the mutex locks that will generate the overhead for each log write API. after all, we can do that but user application must make sure re-initialize the external log system before logging.

Barry-Xu-2018, 2026-01-19:
@fujitatomoya 

`rcl_logging_implementation`  should not be released as a separate package. Like the other packages under `rcl_logging` directory (rcl_logging_interface, rcl_logging_noop, and rcl_logging_spdlog), it should all be included under the rcl_logging target.

https://github.com/ros/rosdistro/blob/master/rolling/distribution.yaml
```yaml
  rcl_logging:
    doc:
      type: git
      url: https://github.com/ros2/rcl_logging.git
      version: rolling
    release:
      packages:
      - rcl_logging_interface
      - rcl_logging_noop
      - rcl_logging_spdlog
      tags:
        release: release/rolling/{package}/{version}
      url: https://github.com/ros2-gbp/rcl_logging-release.git
      version: 3.3.2-1
    source:
      test_pull_requests: true
      type: git
      url: https://github.com/ros2/rcl_logging.git
      version: rolling
    status: maintained
```

Generating Changelog should apply all directories under `rcl_logging`.  For this situation, the manual at `http://docs.ros.org/en/rolling/How-To-Guides/Releasing/Releasing-a-Package.html` does not mention it. I guess it can't be done automatically with the script mentioned in the manual. It needs to be done manually.

```console
root@cscsh-5109201242-dev2:~/ros2_src_rolling/src/ros2/rcl_logging/rcl_logging_implementation# git show 2c98628137643cf126567e582768fd0bd4480894
commit 2c98628137643cf126567e582768fd0bd4480894
Author: Alejandro Hernandez Cordero <ahcorde@gmail.com>
Date:   Fri Oct 17 12:19:31 2025 +0200

    Changelog
    
    Signed-off-by: Alejandro Hernandez Cordero <ahcorde@gmail.com>

diff --git a/rcl_logging_interface/CHANGELOG.rst b/rcl_logging_interface/CHANGELOG.rst
index 8ec5bcf..006816e 100644
    
    Signed-off-by: Alejandro Hernandez Cordero <ahcorde@gmail.com>

diff --git a/rcl_logging_interface/CHANGELOG.rst b/rcl_logging_interface/CHANGELOG.rst
index 8ec5bcf..006816e 100644
--- a/rcl_logging_interface/CHANGELOG.rst
+++ b/rcl_logging_interface/CHANGELOG.rst
@@ -2,6 +2,9 @@
 Changelog for package rcl_logging_interface
 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
 
+3.3.1 (2025-10-17)
+------------------
+
 3.3.0 (2025-04-25)
 ------------------
 
diff --git a/rcl_logging_noop/CHANGELOG.rst b/rcl_logging_noop/CHANGELOG.rst
index 72dc39b..684d5d4 100644
--- a/rcl_logging_noop/CHANGELOG.rst
+++ b/rcl_logging_noop/CHANGELOG.rst
@@ -2,6 +2,14 @@
 Changelog for package rcl_logging_noop
 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
 
+3.3.1 (2025-10-17)
+------------------
+* Cleanup rcl_logging_noop dependencies. (`#132 <https://github.com/ros2/rcl_logging/issues/132>`_)
+  It shouldn't build_export_depend anything (as nothing
+  downstream should link against it), and all of its
+  dependencies can be private.
+* Contributors: Chris Lalancette
+
 3.3.0 (2025-04-25)
 ------------------
 
diff --git a/rcl_logging_spdlog/CHANGELOG.rst b/rcl_logging_spdlog/CHANGELOG.rst
index 805ee7d..a46d6db 100644
--- a/rcl_logging_spdlog/CHANGELOG.rst
+++ b/rcl_logging_spdlog/CHANGELOG.rst
@@ -2,6 +2,9 @@
 Changelog for package rcl_logging_spdlog
 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
 
+3.3.1 (2025-10-17)
+------------------
+
 3.3.0 (2025-04-25)
 ------------------
 * Cleanup overwritten warning messages on error. (`#128 <https://github.com/ros2/rcl_logging/issues/128>`_)
```

fujitatomoya, 2026-01-29:
> Index and Release the new package rcl_logging_implementation.

has been completed with https://github.com/ros/rosdistro/pull/49620

fujitatomoya, 2026-01-29:
@christophebedard yes, please! https://github.com/ros2/rcl/pull/1276 changes dependency on `rcl_logging_implementation`, so rcl also needs to be released. thanks for pointing this out!

christophebedard, 2026-01-29:
done: https://github.com/ros/rosdistro/pull/49638

closed 2026-02-05 by fujitatomoya

fujitatomoya, 2026-02-26:
just FYI.

rcl-logging-implementation package has been released, https://discourse.openrobotics.org/t/new-packages-for-ros-2-rolling-ridley-2026-02-25/52719

```console
root@tomoyafujita-B760M-Pro-RS-D4:~# dpkg -s ros-rolling-rcl-logging-implementation
Package: ros-rolling-rcl-logging-implementation
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 87
Maintainer: Tomoya Fujita <tomoya.fujita825@gmail.com>
Architecture: amd64
Version: 3.4.0-1noble.20260210.194547
Depends: libc6 (>= 2.34), libgcc-s1 (>= 3.0), libstdc++6 (>= 5.2), ros-rolling-rcl-logging-interface, ros-rolling-rcl-logging-spdlog, ros-rolling-rcpputils, ros-rolling-rcutils, ros-rolling-ros-workspace
Description: Provides the ability to load logging implementations at runtime.
 This package acts as an abstraction layer between rcl and the actual logging implementation (e.g., spdlog, noop, or custom implementations).
```