ros2/rcl #1194: Semantics of `rcl_shutdown` and `rcl_context_fini`

fuzzypixelz, 2024-10-28:
I'm opening this issue here as I believe the topic is specific to the `rcl` layer as exposed and documented by this repository. I have a question about the meaning of this doc comment:

https://github.com/ros2/rcl/blob/d9daca7c0af7ecdcd103ece30bb7c64bc919ef24/rcl/include/rcl/context.h#L49-L111

I'm interested in understanding exactly what this means for an rmw implementation such as `rmw_zenoh`, where it's not at all clear what is the expected behavior for the closing of a Zenoh session.

> Invalidation indicates to other entities that the context was shutdown, but is still accessible for use during cleanup of themselves.

**What is the scope of this usage? Is there an exact list of APIs that we expect to be called on an "initialized but invalid" context?**

mjcarroll, 2024-11-08:
Adding to November 12 ROS PMC meeting agenda.

fuzzypixelz, 2024-11-18:
@mjcarroll Thanks for your reply. Do we have any findings from the meeting your mentioned?

fujitatomoya, 2024-11-18:
i only can think of the usage with DDS based RMW implementation.

even after `rcl_shutdown` is called, DDS participants are still active, so that all entities under that participants such as nodes can be still accessible from the application. actually we cannot finalize the context (releasing participants) until all entities are destroyed before, that means it needs to call `rcl_destroy_node` with this invalid context if the node is not destroyed yet.
i think this requirement can be different from implementation details, but it suggests that this could generates undefined behavior.