ros2/rcl #1208: Flaky `test_timer` on Windows

YuanYuYuan, 2025-02-13:
## Description

The flakiness was caught when testing rmw_zenoh. We found that this test sometimes fails even with other RMWs.

The following two tests are flaky.
- TestPreInitTimer.test_timer_info
- TestPreInitTimer.test_timer_info_detect_overrun

## Logs
- Success
```console
❯ C:/dev/ws/build/rcl/test/Release/test_timer.exe --gtest_filter=TestPreInitTimer.test
_timer_info
Running main() from C:\dev\ws-rolling\install\src\gtest_vendor\src\gtest_main.cc
Note: Google Test filter = TestPreInitTimer.test_timer_info
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from TestPreInitTimer
[ RUN      ] TestPreInitTimer.test_timer_info
2025-02-11T07:44:49.624122Z  INFO ThreadId(02) zenoh::net::runtime: Using ZID: 6e804fe
23ec1a7727940cbcd425202cb
2025-02-11T07:44:49.628117Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zeno
h can be reached at: tcp/[::1]:50282
2025-02-11T07:44:49.682638Z  INFO ThreadId(02) zenoh::api::session: close session zid=
6e804fe23ec1a7727940cbcd425202cb
[       OK ] TestPreInitTimer.test_timer_info (63 ms)
[----------] 1 test from TestPreInitTimer (63 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (65 ms total)
[  PASSED  ] 1 test.
```

- Failure: The timestamp difference is about 10ms if the test fails.
```console
❯ C:/dev/ws/build/rcl/test/Release/test_timer.exe --gtest_filter=TestPreInitTimer.test
_timer_info
Running main() from C:\dev\ws-rolling\install\src\gtest_vendor\src\gtest_main.cc
Note: Google Test filter = TestPreInitTimer.test_timer_info
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from TestPreInitTimer
[ RUN      ] TestPreInitTimer.test_timer_info
2025-02-11T07:44:50.322553Z  INFO ThreadId(02) zenoh::net::runtime: Using ZID: bebfee2
cd86cdfed360c3a64beee2cd5
2025-02-11T07:44:50.327007Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zeno
h can be reached at: tcp/[::1]:50284
C:\dev\ws\src\rcl\rcl\test\rcl\test_timer.cpp(1021): error: Expected equality of these
values:
next_expected_call_time
  Which is: 1739259890367916900
call_info.expected_call_time
  Which is: 1739259890377916900

2025-02-11T07:44:50.387349Z  INFO ThreadId(02) zenoh::api::session: close session zid=
bebfee2cd86cdfed360c3a64beee2cd5
[  FAILED  ] TestPreInitTimer.test_timer_info (69 ms)
[----------] 1 test from TestPreInitTimer (70 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (70 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] TestPreInitTimer.test_timer_info
```

## Setup

- Windows 10 VM with [ROS2 rolling installed](https://github.com/ros2/ros2_documentation/pull/4989).
- RCL: https://github.com/ros2/rcl/commit/1427bbb73af6bda69036e5efe1255dfb90329186

fujitatomoya, 2025-02-14:
i cannot reproduce this issue with amd64 linux platform. (`colcon test --event-handlers console_direct+ --retest-until-fail 100 --packages-select rcl --ctest-args -R test_timer`)

i took a quick scan, i guess this is because test time window is too tight for windows.
`timer` could miss the one cycle when it is called, that is why we have exact one cycle off 10ms for the next call.

i think what we can do here is to relieve the time window for the test a bit, like from 10ms to 100ms.
those test will not take much long time or any loops, so execution time should be reasonably quick enough.

fujitatomoya, 2025-02-14:
@YuanYuYuan thanks for creating issue.

could you try https://github.com/ros2/rcl/pull/1209 with your reproducible windows environment?

YuanYuYuan, 2025-02-14:
> [@YuanYuYuan](https://github.com/YuanYuYuan) thanks for creating issue.
> 
> could you try [#1209](https://github.com/ros2/rcl/pull/1209) with your reproducible windows environment?

The failure disappears once the test criterion is relaxed. Thanks @fujitatomoya!

fujitatomoya, 2025-02-14:
@YuanYuYuan great, thanks for testing!

closed 2025-02-14 by ahcorde